OpenKinect/libfreenect2 #886: MAKE error

Abduoit, 2017-07-24:
Hi, 
I am trying to use kinect v2 on fresh machine, I just installed ubuntu 16.04, I am using ROS kinetic, kinect v2, NIVIDIA Quadro K600, CUDA 7.5. 

I ran `cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2` . BUT when I run `make` I get the following error.  Please help me to fix the error?

```
grasp001@grasp001:~/libfreenect2/build$ make
Scanning dependencies of target generate_resources_tool
[  2%] Building CXX object CMakeFiles/generate_resources_tool.dir/tools/generate_resources.cpp.o
[  4%] Linking CXX executable bin/generate_resources_tool
[  4%] Built target generate_resources_tool
[  7%] Building NVCC (Device) object CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o
/home/grasp001/libfreenect2/src/cuda_kde_depth_packet_processor.cu:39:25: fatal error: helper_math.h: No such file or directory
compilation terminated.
CMake Error at cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o.cmake:207 (message):
  Error generating
  /home/grasp001/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o

CMakeFiles/freenect2.dir/build.make:82: recipe for target 'CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o' failed
make[2]: *** [CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o] Error 1
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/freenect2.dir/all' failed
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

xlz, 2017-07-26:
helper_math.h is provided somewhere in your CUDA SDK. You can find it.

Abduoit, 2017-07-26:
Thanks to your reply @xlz 
First, could u plz clarify what do you mean by `helper_math.h` and what should I do ?.
Second, what about the second case, after I removed CUDA, I just want to run `./Protonect` with CPU at least but I get the above error, plz check it out.

xlz, 2017-12-01:
helper_math.h is a file in your CUDA SDK installation. You should find it.

`LIBUSB_ERROR_NO_DEVICE` means your USB driver has some problems. Check https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#linux.

closed 2017-12-01 by xlz

PranjaLBiswas27, 2018-08-22:
Hi,

I have the exact same problem with CUDA 8 and cudnn 6 on Ubuntu 16.04. After reading the comments above, I see that I have helper_math.h file. But what needs to be done to solve the issue. Please let me know. Thanks a lot in advance.

Kind Regards
Pranjal Biswas