OpenKinect/libfreenect2 #887: how can I get world coordinate using depth image?

wuguangbin1230, 2017-08-04:
I calibrated the kinect v2 according to the method described by https://github.com/lakehanne/iai_kinect2/tree/master/kinect2_calibration

And then I got some depth, color images using the following commands:
roslaunch kinect2_bridge kinect2_bridge.launch
roslaunch kinect2_bridge kinect2_bridge.launch
 
The some values of depth image read by matlab are :
0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0
3374	3376	3376	3377	3376	3375	3374	3374
3377	3378	3379	3380	3378	3377	3377	3375
3385	3387	3387	3387	3386	3385	3384	3381
3388	3389	3389	3389	3388	3387	3386	3383
3390	3391	3391	3392	3390	3389	3388	3384
3398	3398	3398	3397	3396	3396	3395	3392
3400	3400	3400	3399	3398	3398	3398	3394
3402	3402	3401	3401	3400	3400	3400	3397
3405	3404	3404	3404	3405	3405	3405	3404
3405	3405	3405	3405	3406	3406	3407	3406
3405	3405	3405	3405	3407	3407	3408	3407
3405	3405	3405	3406	3409	3410	3411	3412
3405	3405	3405	3406	3409	3410	3411	3413
3405	3405	3406	3407	3410	3411	3412	3414
3407	3406	3407	3407	3410	3410	3411	3413
3408	3407	3407	3408	3409	3410	3410	3412
3409	3407	3408	3408	3409	3409	3409	3410

 Does these values are the really distance between object and camera?

wuguangbin1230, 2017-08-04:
And then I got some depth, color images using the following commands:
roslaunch kinect2_bridge kinect2_bridge.launch
rosrun kinect2_viewer kinect2_viewer hd cloud

floe, 2017-08-31:
What, exactly, is your question? Yes, the values are distance in millimeters.

closed 2017-12-01 by xlz