OpenKinect/libfreenect2 #890: How do I improve the OpenGL version ?

wojiaojiao, 2017-08-15:
when i 
$ ./bin/Protonect
show:
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.

$ glxinfo | grep OpenGL
show:
OpenGL version string: 3.0 Mesa 10.5.9

My "OpenGL version string" lower than 3.1.
How do I improve the OpenGL version ?

suijingfeng, 2017-08-16:
glxinfo | grep OpenGL
This is my computer shown:
 
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1060/PCIe/SSE2
OpenGL core profile version string: 4.5.0 NVIDIA 375.66
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.5.0 NVIDIA 375.66
OpenGL shading language version string: 4.50 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 375.66
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

My idea is that if you have a nvidia graph driver card, than you should install nvidia driver carefully,
 then follow the instructions of "https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux", do you run  sudo apt-get install libglfw3-dev ？ what‘s you computer and operate system configuration？

wojiaojiao, 2017-08-17:
@suijingfeng 
My operate system:ubuntu14.04; computer:intel nuc6i7kyk

My computer was originally no problem, but then my computer crash, I into the RecoveryMode repair the ubuntu system.after that, . / Bin / Protonect, is doesn't run, appear the above error!

suijingfeng, 2017-08-17:
@wojiaojiao 
Mesa 10.5.9 is release at July 04, 2015, 
now the latest version is Mesa 17.1.6 which is released at August 7, 2017, try to upgrade it.

In this https://www.mesa3d.org/ page, its say that  
"Mesa 10.5.9 implements the OpenGL 3.3 API, but the version reported by glGetString(GL_VERSION) or glGetIntegerv(GL_MAJOR_VERSION) / glGetIntegerv(GL_MINOR_VERSION) depends on the particular driver being used. Some drivers don't support all the features required in OpenGL 3.3. OpenGL 3.3 is only available if requested at context creation because compatibility contexts are not supported. "

It may be that an old version kernel and/or Mesa is used after you repair the ubuntu system,
try sudo apt-get update then sudo apt-get upgrade,  if it doesn't work.
By the way, install ubuntu16.04 can also be take into consideration.

wojiaojiao, 2017-08-18:
my environment:ubuntu 14.04 LTS,ROS indigo

my solution:

finally i reinstall the system, then according to the following reference to do, it's ok!

Reference resources：https://github.com/OpenKinect/libfreenect2/issues/890

Use the lspci command to find out the model of your graphics card

$ lspci -vnn | grep -i VGA -A 12
because my computer isIntel integrated graphics card，when i $ glxinfo | grep OpenGL, show:The program 'glxinfo' has not been installed.

so Don't try to update the OpenGL version！

if your graphics card is Nvida,and  you just should  update the OpenGL version,Please refer：http://www.binarytides.com/install-nvidia-drivers-ubuntu-14-04/

When you reinstall the system, and install the indigo environment,

do:

mkdir -p ~/catkin_ws/src

cd ~/catkin_ws/src
catkin_init_workspace
cd ..
catkin_make

git clone https://github.com/OpenKinect/libfreenect2.git
cd libfreenect2

cd depends; ./download_debs_trusty.sh
sudo apt-get install build-essential cmake pkg-config

sudo dpkg -i debs/libusb*deb

sudo apt-get install libturbojpeg libjpeg-turbo8-dev
sudo dpkg -i debs/libglfw3*deb
sudo apt-get install -f
sudo apt-get install libgl1-mesa-dri-lts-vivid (If this command conflicts with other packages, don’t do it.)
cd ..
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -DENABLE_CXX11=ON
make
make install
sudo cp ../platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/
Then again plug in kinect2
./bin/Protonect
note:it's ok!then i $ glxinfo | grep OpenGL, show:The program 'glxinfo' has not been installed.

//iai-kinect2
cd ~/catkin_ws/src/
git clone https://github.com/code-iai/iai_kinect2.git
cd iai_kinect2
rosdep install -r --from-paths .
cd ~/catkin_ws
catkin_make -DCMAKE_BUILD_TYPE="Release"
source  ~/catkin_ws/devel/setup.bash(If you have already written to the ~/.bashrc file, you don’t have do it .)
roslaunch kinect2_bridge kinect2_bridge.launch
@suijingfeng

xlz, 2017-09-16:
Closed as there is nothing to fix. Discussion can continue.

closed 2017-09-16 by xlz