OpenKinect/libfreenect2 #894: cmake error building libfreenect2

fma-1724, 2017-08-18:
Hi, i'm following the steps indicated on the README but when i try
 cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2

It gives the following error:
alba.fortes@kinectC2:~/build$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/FREENECT2
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0>=1.0.20'
--   Found libusb-1.0, version 1.0.20
-- Checking for modules 'libva;libva-drm'
--   Found libva, version 0.39.0
--   Found libva-drm, version 0.39.0
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- Checking for module 'glfw3'
--   Found glfw3, version 3.1.2
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "2.0") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Failed
CMake Warning at CMakeLists.txt:309 (MESSAGE):
  Your libOpenCL.so is incompatible with CL/cl.h.  Install ocl-icd-opencl-dev
  to update libOpenCL.so?

CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
CMake Error: File /home/alba.fortes/include/libfreenect2/config.h.in does not exist.
CMake Error at CMakeLists.txt:412 (CONFIGURE_FILE):
  configure_file Problem configuring file

-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 pthread
 va-drm
 va
 /usr/lib/x86_64-linux-gnu/libjpeg.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
CMake Error: File /home/alba.fortes/freenect2.cmake.in does not exist.
CMake Error at CMakeLists.txt:437 (CONFIGURE_FILE):
  configure_file Problem configuring file

CMake Error: File /home/alba.fortes/freenect2Version.cmake.in does not exist.
CMake Error at CMakeLists.txt:438 (CONFIGURE_FILE):
  configure_file Problem configuring file

CMake Error: File /home/alba.fortes/freenect2.pc.in does not exist.
CMake Error at CMakeLists.txt:439 (CONFIGURE_FILE):
  configure_file Problem configuring file

CMake Error at CMakeLists.txt:448 (ADD_SUBDIRECTORY):
  add_subdirectory given source "/home/alba.fortes/doc" which is not an
  existing directory.

-- Configurating examples
CMake Error at CMakeLists.txt:454 (ADD_SUBDIRECTORY):
  add_subdirectory given source "/home/alba.fortes/examples" which is not an
  existing directory.

-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib  
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   OpenCL    yes but buggy
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    yes
--   VideoToolbox    no (Apple only)
-- Configuring incomplete, errors occurred!
See also "/home/alba.fortes/build/CMakeFiles/CMakeOutput.log".
See also "/home/alba.fortes/build/CMakeFiles/CMakeError.log".

alba.fortes@kinectC2:~/build$ make
make: *** No se especifico ningun objetivo y no se encontro ningun makefile.  Alto.

As it shows i tried to do the make as what i really need is only the driver and openni. Any idea how to fix it?

Thanks

xlz, 2017-08-22:
`alba.fortes@kinectC2:~/build$`

This looks wrong. You checked out the code into the wrong place.

closed 2017-08-22 by xlz

fma-1724, 2017-08-22:
thanks for the quick answer.

Could you told me what do you mean by that? How should it look? The build directory is at the same level that the libfreenect2 directory had been automatically created and i have followed the specific commands write on the readme so I'm pretty lose here. Could you help me?

Thanks