OpenKinect/libfreenect2 #896: Cannot build with CUDA-8.0 on Ubuntu 16.04

doc75, 2017-08-23:
Overview Description:

I have built libfreenect properly without CUDA support.
Now I want to build it with CUDA support.
I have downloaded the latest CUDA toolkit from NVIDIA.

Version, Platform, and Hardware Bug Found: 
- Ubuntu 16.04
- master branch of libfreenect2

Steps to Reproduce:

Here is what I did:
```bash
cd libfreenect2
mkdir build
cd build
cmake -DBUILD_EXAMPLES=OFF -DENABLE_CXX11=ON -DBUILD_SHARED_LIBS=OFF ..
make
```

Here the output of **cmake**:
```bash
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Performing Test LIBFREENECT2_THREADING_STDLIB
-- Performing Test LIBFREENECT2_THREADING_STDLIB - Success
-- using stdlib as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0>=1.0.20'
--   Found libusb-1.0, version 1.0.20
-- Checking for modules 'libva;libva-drm'
--   Found libva, version 0.39.0
--   Found libva-drm, version 0.39.0
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- Checking for module 'glfw3'
--   Found glfw3, version 3.1.2
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "2.0") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
[... 285 lines cut ...]

/home/user/Data/Apps/libfreenect2/src/cuda_kde_depth_packet_processor.cu(703): error: too few arguments in function call

/home/user/Data/Apps/libfreenect2/src/cuda_kde_depth_packet_processor.cu(704): error: no suitable conversion function from "const float4" to "float" exists

/home/user/Data/Apps/libfreenect2/src/cuda_kde_depth_packet_processor.cu(704): error: too few arguments in function call

/home/user/Data/Apps/libfreenect2/src/cuda_kde_depth_packet_processor.cu(708): error: too few arguments in function call

Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_00004bc0_00000000-7_cuda_kde_depth_packet_processor.cpp1.ii".
Compilation terminated.
CMake Error at cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o.cmake:266 (message):
  Error generating file
  /home/user/Data/Apps/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o

CMakeFiles/freenect2.dir/build.make:82 : la recette pour la cible « CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o » a échouée
make[2]: *** [CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o] Erreur 1
CMakeFiles/Makefile2:136 : la recette pour la cible « CMakeFiles/freenect2.dir/all » a échouée
make[1]: *** [CMakeFiles/freenect2.dir/all] Erreur 2
Makefile:127 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
```

Any idea of what I did wrong ?

suijingfeng, 2017-08-24:
my computer is also ubuntu16.04 and cuda8.0, but it can build with CUDA support.

lesterlo, 2017-09-04:
Did you install the cuda8.0 patch2 (cuBLAS Patch Update to CUDA 8)?

I have the same problem after I install this cuda patch.

suijingfeng, 2017-09-08:
I'm not install it.

xlz, 2017-09-16:
This

 `/home/user/Data/Apps/libfreenect2/src/cuda_kde_depth_packet_processor.cu(190): error: identifier "dot" is undefined`

dot() is defined in helper_math.h. I keep seeing people having problem with helper_math.h. Does your helper_math.h look ok?

svenevs, 2017-10-25:
> I keep seeing people having problem with helper_math.h. Does your helper_math.h look ok?

I wonder if the samples are just not being installed.  [This is required](https://github.com/OpenKinect/libfreenect2/blob/70df5fa3aed0f007dcf98ba07d0cf55b2a7acd91/CMakeLists.txt#L349-L354) in order to use that stuff (I've never understood why they don't put these in the main `include/` directory...).

@suijingfeng how did you install CUDA?  Was it following the official NVIDIA instructions, or did you use an alternative `apt` package?

Try first finding out where `helper_math.h` is.  The `updatedb` command will take a while if you've never run it before, so be prepared for that.

```console
$ sudo updatedb

$ locate helper_math.h
```

If nothing shows up with `locate`, that means you don't have the CUDA samples installed.  If you do, the next thing to check is make sure that Libfreenect2's build system is including the right directory.    Something like this

```cmake
# FATAL_ERROR is going to stop CMake immediately, does this directory
# exist and/or is it the same as `locate helper_math.h`?
message(FATAL_ERROR "Samples Include: ${CUDA_TOOLKIT_ROOT_DIR}/samples/common/inc")

# Just before the part that does this:
CUDA_INCLUDE_DIRECTORIES(
  "${MY_DIR}/include/"
  "${CUDA_TOOLKIT_ROOT_DIR}/samples/common/inc"
  "${NVCUDASAMPLES_ROOT}/common/inc"
  "${NVCUDASAMPLES8_0_ROOT}/common/inc"
)
```

closed 2018-11-18 by xlz

griwes, 2019-10-03:
> I've never understood why they don't put these in the main `include/` directory...

I have it on good authority that it's probably because... not a part of the toolkit. Samples are _not_ actually a part of the toolkit. If you want to use a feature from outside of the toolkit, you should detect it separately... and preferably just don't depend on a header file that is not installed when a user makes a (very reasonable) decision not to install the samples.

xlz, 2019-10-03:
I just checked my CUDA installation,
```
$ dpkg -S helper_math.h
cuda-samples-10-0: /usr/local/cuda-10.0/samples/common/inc/helper_math.h
```
So helper_math.h is from the cuda-samples-10-0 package. And cuda-samples-10-0 is in fact a hard dependency of `cuda-toolkit-10-0`. So in my installation it's actually impossible to install CUDA without samples.

Morwenn, 2019-10-03:
I used a runfile downloaded from [this page][1] to install the toolkit and installed it without the samples, which is when I ran into this issue.

  [1]: https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&target_distro=Ubuntu&target_version=1804

xlz, 2019-10-03:
`helper_math.h` is not redistributable, so until the part of helper_math.h gets reimplemented here you have to install the samples or just use the debs to install cuda.

Morwenn, 2019-10-03:
Oh, I know how to fix/circumvent the issue, I'm just backing up the claim that there are reasonable situations where it can happen :)