OpenKinect/libfreenect2 #898: OpenNI2 depth always runs at 15fps

mediadog, 2017-08-26:
I am using the 0.2.0 x64 libusbK binaries under Win7 x64.  The depth stream in my OpenNI2 app always runs at exactly 15fps.  This is odd, because in reading the source it appears the libfreenect2 OpenNI driver always sets it to 30fps no matter what is passed in to the OpenNI call.

The fact that it is exactly 15fps, and happens on a laptop as well as a high-end gaming system makes me doubt it is any kind of throughput issue, but rather a setting somewhere.  I am only opening the depth stream, not the IR or RGB (at least via the OpenNI2 calls, no idea if they are getting opened by libfreenect2 anyway).

And even more odd, two Kinect2s on the same USB3 adapter will run, both at 15fps.  And I have tried unplugging all USB3 devices from the PC except the one K2.

Any ideas greatly appreciated!

mediadog, 2017-08-27:
And this same program can happily set Primesense/Xtion cameras to 30 or 60fps, so that code is not the problem.  And tested it on a third system with the same 15fps result.

mediadog, 2017-08-31:
This seems to be a result of the RGB stream throttling down due to low light levels, as if I add light to the scene the depth frame rate jumps to 30fps.  But since I am never opening the RGB stream, only the depth stream, this should not be happening, yes?  It does not do this with the MS SDK.

JoshBlake, 2017-08-31:
Auto-throttling of the color stream from 30 FPS to 15 FPS is a Kinect firmware feature (to maintain properly exposed color image) and so was experienced with the Kinect SDK as well. See this thread on it: https://social.msdn.microsoft.com/Forums/en-US/9e81bae1-2b3a-4926-9e44-b2f400ccb312/color-only-15fps?forum=k4wv2devpreview

In your scenario, if the depth stream is waiting for matching color frames then it will also be throttled. To get max FPS from each stream one would have to decouple the streams.

mediadog, 2017-09-01:
OK, but as I said I am only opening the depth stream, so I would think there would be no coupling.  Is libfreenect2 or the libfreenect2 OpenNI2 driver opening both (or all three) streams in all cases and just transferring data for the "open" one?  Is this decoupling something that would need to happen in the libfreenect2 internals?  I would have thought this kind of coupled behavior would only happen if depth and RGB sync is set, which I am explicitly turning off at the OpenNI2 level.

mediadog, 2017-09-01:
Hmmmm, in the file src\OpenNI2\DeviceDriver.cpp I found:

    /* todo: for DriverBase
    virtual void* enableFrameSync(oni::driver::StreamBase** pStreams, int streamCount);
    virtual void disableFrameSync(void* frameSyncGroup);
    */

Which makes me think that libfreenect2 is syncing the streams by default.  How to change this, or add these calls?

xlz, 2017-09-16:
See #738. Close as duplicate.

closed 2017-09-16 by xlz