OpenKinect/libfreenect2 #899: No docs in binary release

mediadog, 2017-08-26:
I can find no documentation in the binary release libfreenect2-0.2.0-libusbk-vs2015-x64.  The source distribution has the doc sources.  The created doc should be included in the binary release.

In particular I am looking for the possible values of the LIBFREENECT2 environment variables.

Thanks!

floe, 2017-08-31:
https://openkinect.github.io/libfreenect2/index.html

mediadog, 2017-09-04:
Thanks for the pointer floe.  But those docs are not part of the release package, which they should be.

And you'll notice I said "I am looking for the possible values of the LIBFREENECT2 environment variables", and the docs at the link simply say:

> LIBFREENECT2_LOGGER_LEVEL: The default logging level if not explicitly set by the code. LIBFREENECT2_PIPELINE: The default pipeline if not explicitly set by the code. LIBFREENECT2_RGB_TRANSFER_SIZE, LIBFREENECT2_RGB_TRANSFERS, LIBFREENECT2_IR_PACKETS, LIBFREENECT2_IR_TRANSFERS: Tuning the USB buffer sizes. Use only if you know what you are doing.

There is no hint as to what the possible values are and why you would want to set them.

floe, 2017-09-04:
OK, agreed, this could be documented better. Right now, you'll probably need to search the code to see where the envvars are used...

xlz, 2017-09-16:
I agree generally binary releases include documentation to enable offline development. But I don't think anyone is using the binary release for offline development. The build itself doesn't turn on all the features. Everyone builds from source. So it's kind of nobody-cares issue. I'll keep it in mind next time I make a release.

mediadog, 2017-09-16:
Well, I do not build from source, and I care, so odds are there are others that do too - or would, but are put off by the lack of ease-of-use.  The build-from-source approach is understandable during the hacking/development phase, but for a tool usable by a general audience it needs to be a package they can just download, learn, and use.  Libfreenect2 is _almost_ there!  I'd be happy to help improve the docs, but I have to figure out how it works first...  Thanks!