OpenKinect/libfreenect2 #901: Kinect USB adapter wiring

scj643, 2017-08-30:
I'm trying to find some documentation so I can DIY a USB adapter.

floe, 2017-08-31:
Google is your friend, but I'll post it here for reference anyway: http://www.tatsch.it/kinect-2-usb-3-modification/
Please note that this may have some side effects regarding USB3 compatibility, no guarantees at all.

scj643, 2017-09-01:
I saw that before but what I mean is if I splice the connection cable what wires go where.

floe, 2017-09-04:
If you're trying to split open the original cable and add a power connection (and probably a new USB3 plug), I'd recommend against that, because the shielding will very likely be damaged and you'll get even more USB3 issues. If you really want to get by without the adapter box, open the Kinect itself and solder to the internal power connections.

suijingfeng, 2017-09-06:
https://openkinect.org/wiki/Kinect_2 ，this link maybe helpful