OpenKinect/libfreenect2 #907: Issue with reconstructing libfreenect2 frames

sjdrc, 2017-09-07:
I want to convert a cv::Mat back into a libfreenect2::Frame.

```
listener_.waitForNewFrame(frames_);
libfreenect2::Frame *rgb = frames_[libfreenect2::Frame::Color];
libfreenect2::Frame *depth = frames_[libfreenect2::Frame::Depth];
cv::Mat color_tmp(rgb->height, rgb->width, CV_8UC4, rgb->data);
cv::Mat depth_tmp(depth->height, depth->width, CV_32FC1, depth->data);
		
libfreenect2::Frame *rgb2 = new libfreenect2::Frame(color_tmp.cols, color_tmp.rows, 4, color_tmp.data);
libfreenect2::Frame *depth2 = new libfreenect2::Frame(depth_tmp.cols, depth_tmp.rows, 1, depth_tmp.data);
cv::Mat color_tmp2(rgb2->height, rgb2->width, CV_8UC4, rgb2->data);
cv::Mat depth_tmp2(depth2->height, depth2->width, CV_32FC1, depth2->data);
```

color_tmp and color_tmp2 both look the same when visualised with cv::imshow. If I call registration->apply on rgb and depth, I get the expected result. However when I used rgb2 and depth2, the resultant registered image is entirely black. Where am I going wrong here?

suijingfeng, 2017-09-07:
@sjdrc 

The third parameters of Frame construction is Bytes per pixel, try replace 1 with 4. 

[](https://github.com/OpenKinect/libfreenect2/blob/master/include/libfreenect2/frame_listener.hpp)
` /** Construct a new frame.
   * @param width Width in pixel
   * @param height Height in pixel
   * @param bytes_per_pixel Bytes per pixel
   * @param data_ Memory to store frame data. If `NULL`, new memory is allocated.
   */
  Frame(size_t width, size_t height, size_t bytes_per_pixel, unsigned char *data_ = NULL);`

floe, 2017-10-13:
Not really a libfreenect issue?

closed 2017-10-13 by floe