OpenKinect/libfreenect2 #908: wrong jpeg library version: library is 90, caller expects 80

walterwang, 2017-09-10:
os: ubuntu 16.04
I have anaconda3: 
Steps to Reproduce:

1. when I run ./bin/Protonect I get an error: 
Wrong JPEG library version: library is 90, caller expects 80

during build: 
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 pthread
 va-drm
 va
 /home/walter/anaconda3/lib/libjpeg.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so

But conda's libjpeg is already version 9.2

walter@walter-desktop:~/anaconda3/lib$ ls -la | grep libjpeg
-rw-r--r--  2 walter walter  2078420 Apr 15  2016 libjpeg.a
-rwxr-xr-x  1 walter walter      929 Sep 10 04:04 libjpeg.la
lrwxrwxrwx  1 walter walter       16 Sep 10 04:04 libjpeg.so -> libjpeg.so.9.2.0
lrwxrwxrwx  1 walter walter       16 Sep 10 04:04 libjpeg.so.9 -> libjpeg.so.9.2.0
-rwxr-xr-x  2 walter walter  1073308 Apr 15  2016 libjpeg.so.9.2.0

walterwang, 2017-09-10:
fixed by downgrading the libjpeg version in conda from 9.2 to 8

closed 2017-09-10 by walterwang

VCBE123, 2018-05-28:
I can't figure out how to solve this probolem