OpenKinect/libfreenect2 #909: Depth image limited to 5 Meters vs. MS SDK of 8 Meters

mediadog, 2017-09-19:
The depth image I am getting back cuts off at 5 meters, while the MS SDK depth image goes out to 8 meters.  Is this a runtime setting somewhere?

xlz, 2017-09-20:
Freenect2Device::setConfiguration. It's not exposed through OpenNI2. You can patch src/openni2/DeviceDriver.cpp:setFreenect2Device() for this functionality.

mediadog, 2017-09-20:
OK, thanks.  But shouldn't the Freenect2  default be 8 meters (or its max depth capability, if not 8M).  Where did 5M come from?

xlz, 2017-09-20:
https://social.msdn.microsoft.com/Forums/en-US/c95d3e40-6ed6-47a1-a206-5ff26c889c29/kinect-v2-maximum-range

closed 2017-09-20 by xlz

mediadog, 2017-09-20:
This should not be closed!  They are talking about maximum skeleton tracking depth, an entirely different thing.  The depth image returned by default should be the maximum depth the camera is capable of - what is the sense of limiting it to the skeleton depth, which libfreenect2 does not even do?   In their answer they even say "The Kinect v2 can physically sense depth at 8 meters."

So when returning a DEPTH image, the maximum DEPTH distance should be used by default, not the maximum SKELETON distance.  That is how every depth camera I have worked with works, including the Kinect v1 and v2, with their native SDK and via OpenNI.  It is a flaw that libfreenect2 does not do the same thing.  Thanks!

xlz, 2017-09-20:
The link explains the reason. If you bother to look it up, the official default maximum range of Kinect v2 is 4.5m. https://developer.microsoft.com/en-us/windows/kinect/hardware

The theoretical maximum is 18.75m. But we will not use either of those. The purpose of the default is to have *good* data, in the principle of least surprise. Data with more than 4.5m has increased error, which is not obvious. The default doesn't stop you from using other maxima but the default is sane.

mediadog, 2017-09-21:
Well, then why when I fire up good old MS Kinect Studio does the depth image and pointcloud go out to - you guessed it - 8 meters?  And when I use the 2.0 SDK to open a Kinect it works out to - yup - 8 meters.  And note these threads:

https://social.msdn.microsoft.com/Forums/vstudio/en-US/bfa151bd-4d17-416c-b105-c76b1ec051c9/how-can-i-set-the-threshold-of-max-depth-kinect-v2?forum=kinectv2sdk

https://social.msdn.microsoft.com/Forums/en-US/9758c07e-4d28-4942-bd76-30ca09d17c8d/modify-minmax-range-of-kinect-2?forum=kinectv2sdk

"Not at this time. SDK 2.0 with Kinect v2 has the 0.5-8m fixed range."

And since I can find no case where the depth image is limited to less than 8M, having data just dropped I get everywhere else was a pretty big surprise!