OpenKinect/libfreenect2 #912: Registered Color Image and Undistorted Depth are Black

wired365, 2017-09-21:
I've been using the library without issues for months now and just recently I ran into an issue where the registered color image appears to be set to all max values (black) and the undistorted depth image is all zeros. The distorted depth image and the IR image appear to be populated correctly. They look correct when debugging and appear visually correct when saved as a TIFF.

When I step through the code the issue appears to come from somewhere within `Registration.apply()`. The issue, I think, is coming from the `distort_map` and subsequently `map_dist` being set entirely to -1. I wasn't able to determine where the -1 value gets set or where it would be overwritten. From what I understand of the code, because all of the values are determine to be "outside of the depth image" the depth image simply gets set to 0 and later the registered image gets set to black since there is no non-zero depth value.

Has anyone ever run into this issue or know what could be causing it?

I'm running on a Windows 10 machine with only 1 Kinect V2 currently connected.

wired365, 2017-09-21:
I dug a little bit deeper and found that the `IrCameraParams` and `ColorCameraParams` used in the constructor of `RegistrationImpl::RegistrationImpl` appear to all be set to -431602080, which means they are not being correctly set somewhere previously. This appears to be causing the issue in the constructor which sets the map_dist to -1 causing my original issue.

Any suggestions as to why these parameters wouldn't be set properly?

wired365, 2017-09-21:
It appears this might be a USB driver issue somehow. I'm still attempting various troubleshooting attempts to resolve it, no luck so far. There was an update to Windows 10 recently that seems to have messed with the USB communication that sends and receives the camera parameters. The raw data output appears to still be functional but anything that utilizes the camera parameters, such as registration, doesn't work.

I've test on 2 different Windows 10 machines with known stable and working copies of my code, everything has either timed out via a LIBUSB_TIMEOUT_ERROR or the registered images simply come out black due to the camera parameters being invalid.

xlz, 2017-09-21:
If distorted depth images are fine it means IrCameraParams should be ok.

You should use Freenect2Device::getColorCameraParams,getIrCameraParams to get and print the parameters right after Freenect2Device::start(), which is an early point. See what exactly is not right.

LIBUSB_TIMEOUT_ERROR looks to be a separate thing. I can only suggest trying usbdk vs libusbk.

wired365, 2017-09-21:
Thanks @xlz I am currently using usbdk, version 1.0.19. The "LIBUSB_TIMEOUT_ERROR" appears to be a one time thing and has since gone away.

I'm confused, how does the IrCameraParams relate to the distorted depth. For clarity I mean the raw depth image that has not be processed for distortion is what is correct. The depth image that comes out of the `registration->apply()` is what appears to be all 0's.

In any case I called `Freenect2Device::getColorCameraParams` and `Freenect2Device::getIrCameraParams` and in both cases the output is a parameter object in which all of the fields are -431602080. I made the call immediately after the device was opened.

xlz, 2017-09-21:
IrCameraTables in libfreenect2.cpp are derived from IrCameraParams. You should post the full logs if you can. There should be some "pixels in x/ztable have incorrect undistortion."

-431602080.0 in float is 0xcdcdcdcd which is uninitialized memory value from VC++.

What about examining `result` above https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L729?

wired365, 2017-09-21:
Sorry, can you refresh my memory as to how to post the full logs or which you are referring to?

what should I be looking for with result? It appears to be populated with a variety of numbers on the order of 154, 249, 182, with zeros throughout and 249 being the largest.

xlz, 2017-09-21:
The full logs - what is printed in the console.

Ok, about result, use something like
```
for (unsigned char c : result) {
  printf("%02x ", (int)c);
}
```

wired365, 2017-09-21:
Oh that's what you meant, sorry I thought you meant something more detailed.

Below is the full output with the print statement added:

```
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
ab 36 b6 43 ab 36 b6 43 00 00 00 00 1e 86 7f 43 b2 0e 4b 43 fc be c6 3d 9f 5e 8c be 00 00 00 00 00 00 00 00 4a d0 be 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] already stopped, doing nothing
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```

xlz, 2017-09-21:
The USB data looks quite alright.
```
ab 36 b6 43 -> 364.427094
ab 36 b6 43 -> 364.427094
1e 86 7f 43 -> 255.523895
b2 0e 4b 43 -> 203.057404
fc be c6 3d -> 0.097044
9f 5e 8c be -> -0.274159
```
It should be just invalid memory access somewhere. Immediately follows the above is Freenect2DeviceImpl::setIrCameraParams. Maybe you can poke around it. Maybe you can Visual Studio to set some watchpoints with ir_camera_params_ in Freenect2DeviceImpl.

wired365, 2017-09-22:
After lots of debugging I found the issue was in my code. At some point in refactoring or updating my code I moved the creation of the registration object to a location before the device is started causing the registration to be initialized with uninitialized memory. I moved the registration creation to after the start call command and everything is fixed.

Sorry for wasting your time with such a simple mistake. Thanks again for all of the help.

closed 2017-09-27 by xlz