OpenKinect/libfreenect2 #913: bigdepth to point cloud

madsherlock, 2017-09-26:
Overview Description:

Hello,
I'm trying to get 3D point cloud data using the bigdepth image from Registration->apply().
Is there an example on how to accomplish this?
I can convert "smalldepth" and "registered rgb" images to a point cloud using depth camera intrinsics, but I seem to get incorrect world coordinates in my attempt with bigdepth and the original rgb image.

I would like to follow the simple formulas of:
z = bigdepth / 1000  // assuming top and bottom row already removed
x = ([1,...,1920] - intrinsics.cx)/intrinsics.fx .* z
y = ([1,...,1080] - intrinsics.cy)/intrinsics.fy .* z

I'm using the depth camera intrinsics right now, and obviously it's not working, what with neither x nor y axis being centered around zero.

Essentially, I'm looking for a way to convert 1920x1080 RGB-D from libfreenect2 to a PCL pointcloud.

Additional Information:

By point cloud I mean a cloud of 1920x1080 points in the world coordinate system (x,y,z), with color information (r,g,b).

floe, 2017-10-13:
Hm, I think we never actually had that use case. The bigdepth image is mostly for convenience and doesn't really contain more information than the rectified depth image itself, every pixel in bigdepth is part of a 5x3 patch that shares the same depth value and corresponds to a single pixel in the regular depth image. So your point cloud would consist of rectangular patches anyway... not sure if this is what you'd want?

madsherlock, 2017-10-16:
Thanks for your response. I've "sort of" solved my issue today, but would like to explain the reasons for my asking here.

First, the "solution":
I was creating the point cloud correctly. My observed distortions were due to lack of calibration and due to reflective properties of my scene. So, I still have my problem, but it's not in the creation of the point cloud.

I have two use cases, and I think you would also find them relevant:

**Use case 1**:
I have three different object types of the same shape but different texture. All are small and detailed.
At sufficient resolution, all object types can be detected reliably in 2D.
512x424 is insufficient resolution for this, whereas 1920x1080 is very good.
I need the 3D position of my objects.

I can find the objects in 2D (Full HD), and would like to be able to quickly obtain their 3D position.

libfreenect2 provides much functionality here, in that I can look up a depth value in "bigdepth", and convert this to XYZ in millimetres by use of the RGB camera intrinsics.

The problem is extrinsic calibration: I cannot perform this, due to the unconventional registration method.

**Use case 2**:
I want to do surface reconstruction with detailed RGB information. This is where a HD point cloud would be nice.

Again, the problem is extrinsic calibration.

--------------------------

My new problem:
As far as I can see, the "bigdepth" filter of libfreenect2 is one of the best methods for depth upsampling. However, it works in the RGB frame registered by an unconventional method.

Is it possible to apply the "bigdepth" filter algorithm to an image pair registered by an alternative method?
Eg. I make my own mapping table using an extrinsic matrix, and apply the "bigdepth" filter using this?
Is the "bigdepth" filter algorithm described in a paper or documented somewhere?

djivani, 2017-11-14:
@madsherlock How did you go about obtaining Euclidean (X,Y,Z) coordinates from depth or undistorted depth images, assuming you have the IR Camera Parameters (cx,cy,fx,fy)?
Does the approach you mention work directly with depth images? If yes, are the results you obtain same as the ones obtained by `registration->getPointXYZ()`?
>I would like to follow the simple formulas of:
z = bigdepth / 1000 // assuming top and bottom row already removed
x = ([0,...,1919] - intrinsics.cx)/intrinsics.fx .* z
y = ([0,...,1079] - intrinsics.cy)/intrinsics.fy .* z

floe, 2017-11-15:
@madsherlock sorry for the long delay. The actual upsampling filter can be found here:
https://github.com/OpenKinect/libfreenect2/blob/master/src/registration.cpp#L220-L234

xlz, 2017-12-01:
@madsherlock The "unconventional registration method" was reverse engineered from Microsoft SDK. It's used for the factory parameters. People tend to do their own registration with their own extrinsic calibration methods so we didn't put it a way to customize extrinsic transformation.

The "bigdepth" filter you asked for was originally derived from this OpenCV test example (first half): https://github.com/xlz/libfreenect2/commit/a0c0b6e5402ad363dcd5bbe58ebab8d093ea2f80. I can describe it:
```
bigdepth = 1920x1080 pixels of 65536.0
for each (x,y) in 512x424 depth image:
    find the registered (cx,cy) in 1920x1080 color image
    assign smalldepth[x,y] to bigdepth[cx,cy] if smaller
// bigdepth will have many holes
Erode bigdepth by 5,3 so holes are filled.
```
Hopefully this will be easier to implement with OpenCV in you custom code.

closed 2017-12-01 by xlz