OpenKinect/libfreenect2 #916: Kinect device not detected in custom code, but works in Protonect

anuppari, 2017-10-12:
Overview Description:
I'm trying to replicate a basic version of Protonect in Visual studio 2015. The protonect executable works fine, but my code shows no devices connected. A weird symptom is that when protonect runs, it detects 6 usb devices, and 1 kinect device. but my code detects 8 usb devices and no kinect.

I've stripped my code to the basic form to replicate the error:

```
#include <iostream>

#define GLFW_DLL
#include <libfreenect2/libfreenect2.hpp>
#include <libfreenect2/frame_listener_impl.h>
#include <libfreenect2/packet_pipeline.h>

int main(int argc, char *argv[])
{
  // Initialize freenect device (kinect), pipeline and listener
  std::cout << "here1" << std::endl;
  libfreenect2::Freenect2 freenect2;

  if (freenect2.enumerateDevices() == 0)
  {
    std::cout << "no device connected!" << std::endl;
    return -1;
  }
return 0;
}
```

Version, Platform, and Hardware Bug Found:

1. Version: https://github.com/OpenKinect/libfreenect2/commit/93769abc75654888a6848e2520c50f1f952ce72e
2. Platform: Windows 10 64bit
3. Using UsbDk and "WDF KinectSensor Interface 0" shows in device manager

Steps to Reproduce:

1. Build a basic example of enumerating devices

Actual Results:
No kinect detected.

Expected Results:
1 kinect detected.

Reproducibility:

Additional Information:
My code compiles and links fine

anuppari, 2017-10-12:
So I tried running my code without the kinect plugged in and I don't get the LIBUSB_ERROR_NOT_FOUND error, I only get `found 0 devices`, which matches the protonect error when the kinect is not plugged in. Also, my code detects 5 usb devices when kinect not plugged in, protonect detects 3, so both see the kinect as 3 usb devices.

On the other hand, when the kinect is plugged in, my code outputs:
```
8 usb devices connected
failed to open Kinect v2: @1:29 LIBUSB_ERROR_NOT_FOUND Entity not found
```
So I guess there's some issue with opening the kinect. Still weird that protonect works fine.

floe, 2017-10-13:
Hm, that may indeed be a bug. What happens if you create a pipeline before calling `enumerateDevices()`, e.g. by calling `pipeline = new libfreenect2::CpuPacketPipeline();` first?

anuppari, 2017-10-13:
So I figured it out: long story short, the wrong libusb dll was being called. I have a libusb install that's on the system path for use with PCL, which doesn't use UsbDk. During building of libfreenect and protonect, all the correct dlls get put in the same directory as the protonect binary, which I guess windows prioritizes over the libusb on the system path. Copying the libusb dll from the libfreenect bin directory over to the working directory of my code fixes it and everything works.

So now that usbdk is installed, should I just use the usbdk version of libusb everywhere (even in PCL)? What is usbdk doing beyond plain libusb? When my code was calling vanilla libusb in PCL, there was no crash, so it looks like all the library functions are the same in both versions, they just perform differently when usbdk is installed.

xlz, 2017-10-24:
@anuppari Are you using PCL to access other USB devices? UsbDk is for isochronous USB transfers used by Kinect 2. If you're not using PCL to access other USB devices you can probably ignore the libusb version used by it.

closed 2017-10-24 by xlz