OpenKinect/libfreenect2 #917: No rule to make target `install-openni2'

amozooz, 2017-10-16:
Hi,
I was trying to follow the installation instruction for Mac. Everything was fine, I successfully finished this part "Run the test program: ./bin/Protonect"

However, I stuck on the following part:
Test OpenNI2. make install-openni2 (may need sudo), then run NiViewer. Environment variable LIBFREENECT2_PIPELINE can be set to cl, cuda, etc to specify the pipeline.

`cindydeMacBook-Pro:build cindy$ sudo make install-openni2
make: *** No rule to make target `install-openni2'.  Stop.`

Is there anyone know what is going on?
Thank you so much!!
 
Version, Platform, and Hardware Bug Found:

hitcww, 2017-10-18:
I have the same problem!

svenevs, 2017-10-25:
It wasn't configured correctly.  Something is wrong with the packaging of / finding of OpenNI2 on OSX.  With a `brew` installed `openni2`, you get the following output:

```
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
```

So the way we got there was something like

```cmake
$ cd /path/to/libfreenect2
$ mkdir build
$ cd build
$ cmake ..
```

Now what you will want to do is run `ccmake`.

### Step 1: find `libOpenNI2.dylib` manually

For me, Libfreenect2 is able to find the right include directory, but not the right library.  The right library is strangely named.  On my system, the full path was `/usr/local/Cellar/openni2/2.2.0-debian/lib/ni2/libOpenNI2.dylib`.  Go hunt around in `/usr/local/Cellar/openni2` and figure out where it is.

### Step 2: specify this manually

```
# After having run `cmake ..`, still in the `build` directory
$ ccmake .
```

So when the terminal gui opens, use the down arrow key and get to the OpenNI2 section, should be near the bottom.  For reference, right now mine looks like this:

```
OpenNI2_INCLUDE_DIRS             /usr/local/include/ni2
OpenNI2_LIBRARY                  OpenNI2_LIBRARY-NOTFOUND
```

When you are over the `OpenNI2_LIBRARY`, hit `enter` to edit it.  Paste the full path to `libOpenNI2.dylib` and then hit `enter` again.  So now mine looks like this:

```
OpenNI2_INCLUDE_DIRS             /usr/local/include/ni2
OpenNI2_LIBRARY                  /usr/local/Cellar/openni2/2.2.0-debian/lib/ni2/libOpenNI2.dylib
```

Now hit the `c` key, which will re-run CMake.  You should now be able to hit the `g` key to re-generate the `Makefile`.  After that, `q` for exit.

### Alternative

The `ccmake` gui is kind of hard to navigate if you haven't used it before.  You should also be able to do

```
$ cmake .. -DOpenNI2_LIBRARY="/usr/local/Cellar/openni2/2.2.0-debian/lib/ni2/libOpenNI2.dylib"
```

### Should be ready

Now build

```
$ make -j 4

$ ls lib
libfreenect2-openni2.0.dylib* libfreenect2-openni2.dylib@   libfreenect2.0.2.0.dylib*     libfreenect2.0.2.dylib@       libfreenect2.dylib@
```

If you don't see `libfreenect2-openni2.0.dylib` then `make openni-install` will not work, because it isn't building the openni2 bindings.

I hope that was more helpful than confusing!

closed 2017-10-25 by xlz

amozooz, 2017-10-26:
Hi @svenevs 
Thanks for the reply. I think this is exactly my problem.

However, after I found and changed the path of OpenNI2_LIBRARY, and hit "c" key, it shows
` CUDA_TOOLKIT_ROOT_DIR not found or specified`

then I tried with "e" key to exit and the "g" key to generate
and got 
```
CMake Warning (dev):
   Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
   --help-policy CMP0042" for policy details.  Use the cmake_policy command to
   set the policy and suppress this warning.

   MACOSX_RPATH is not specified for the following targets:

    freenect2
    freenect2-openni2

 This warning is for project developers.  Use -Wno-dev to suppress it.
```
Do you know how to fix it?

Thank you so much!

xlz, 2017-10-26:
You can ignore the warnings. They are for project developers.

amozooz, 2017-10-26:
I see. Thanks for the reply!

PR #922 (2017-10-25): add osx library path for brew install
Fixes #917.

It seems the `openni2` brew package installs to [`/usr/local/lib/ni2`](https://github.com/Homebrew/homebrew-science/blob/58e8834dd8e916f51a26e96de1573018a82914ca/openni2.rb#L41).  The driver libraries are in `/usr/local/lib/ni2/OpenNI2/Drivers`, but I don't think that matters.