OpenKinect/libfreenect2 #921: Installed OpenKinect on Ubuntu 14.04, but cannot find viewer

Kayee6, 2017-10-24:
I am using Ubuntu 14.04 and installed libfreenect2 following the instruction from here: 
[https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux](https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux)
But before that, I also installed libfreenect from 'Get started OpenKinect' [https://openkinect.org/wiki/Getting_Started](https://openkinect.org/wiki/Getting_Started)  

So currently I have two files on my ubuntu: 
```
Desktop           freenect2      Music                    Pictures    Videos
Documents         libcudnn.so.6  netcdf-3.6.3             Public
Downloads         libfreenect    netcdf-3.6.3.tar         Templates
examples.desktop  libfreenect2   NVIDIA_CUDA-8.0_Samples  tensorflow
```
After installing libfreenect2, I tried `./bin/Protonect` , 
but here shows some weird information:
```

kj@kj-Alienware-Area-51-R2:~/libfreenect2/build$ ./bin/Protonect
./bin/Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/kj/libfreenect2/build/lib/libfreenect2.so.0.2)
./bin/Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/kj/libfreenect2/build/lib/libfreenect2.so.0.2)
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 002590564047
[Info] [Freenect2Impl] found 1 devices
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce GTX 1080 @ 1733MHz Memory 8112MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 002590564047
device firmware: 4.0.3916.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 960
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Debug] [RgbPacketStreamParser] skipping rgb packet!

...(blabla)

The viewer is turned off. Received 2700 frames. Ctrl-C to stop.
The viewer is turned off. Received 2800 frames. Ctrl-C to stop.
^C[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

```

It seems the viewer is not opened. But is it opened automatically? Or I need to install other app to run it successfully? I am really new to both using Ubuntu and Kinect, sorry for if I asked some strange question. I greatly appreciate someone could answer my question!

svenevs, 2017-10-25:
Welcome to linux :smile:  It seems you did not have `glfw` setup correctly, the program output you pasted is what happens when the viewing application component is not built.  When you run CMake, it'll output at the end what it is actually going to do.  Something like this:

```
-- Feature list:
--   CUDA    yes
--   CXX11    yes
--   Examples    yes
--   FLAG_SEARCH_PATHS_FIRST    1
--   OpenCL    disabled
vvvvvvvvvvvvvvvvvvvv
--   OpenGL    yes <
^^^^^^^^^^^^^^^^^^^^
--   OpenNI2    disabled
--   TegraJPEG    disabled
--   Threading    std::thread
--   TurboJPEG    no
--   VAAPI    disabled
--   VideoToolbox    yes
```

If you got `OpenGL no`, then it did not find everything it needed for building OpenGL.  Until it says `OpenGL yes`, the viewer will not be built.  Revisit the install instructions you linked and make sure you have GLFW3 setup.  Hope that helps some!

Kayee6, 2017-10-25:
Thank you for the help! I checked the version of my OpenGL by using `glxinfo | grep "OpenGL version"`, and got `OpenGL version string: 4.5.0 NVIDIA 375.66`. And I confirmed `OpenGL    no`.

Then I try to install GLFW3 again by the comand `sudo apt-add-repository ppa:floe/beignet; sudo apt-get update; sudo apt-get install beignet-dev; sudo dpkg -i debs/ocl-icd*deb` from the official link and got follows:
```
dpkg: error processing archive debs/libglfw3*deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 debs/libglfw3*deb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libxcursor-dev libxinerama-dev linux-image-4.4.0-31-generic
  linux-image-extra-4.4.0-31-generic x11proto-xinerama-dev
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 40 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libgl1-mesa-dri-lts-vivid

```
 I saw someone saying that libglfw3 is only available for ubuntu 14.10 and later, but I have 14.04 on my computer. Also I tried to compile it from the source, but again, I got this:
```
-- Could NOT find Vulkan (missing:  VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Using X11 for window creation
CMake Error at CMakeLists.txt:248 (message):
  The RandR headers were not found
-- Configuring incomplete, errors occurred!
See also "/home/kj/glfw-master/CMakeFiles/CMakeOutput.log".
See also "/home/kj/glfw-master/CMakeFiles/CMakeError.log".

```

Then by chance, I just used `sudo apt-get update && sudo apt-get install libglfw-dev`, it seems it worked for installing something:
```
Ign file:  InRelease
Get:1 file:  Release.gpg [819 B]                                               
Get:2 file:  Release [574 B]                                                   
Hit http://storage.googleapis.com stable InRelease                             
Ign http://dl.google.com stable InRelease                                      
Ign http://us.archive.ubuntu.com trusty InRelease                              
Ign file:  Translation-en_US                                                   
Ign file:  Translation-en                                                      
Hit http://us.archive.ubuntu.com trusty-backports InRelease                    
Get:3 http://dl.google.com stable Release.gpg [819 B]                          
Hit http://storage.googleapis.com stable/jdk1.8 amd64 Packages                 
Hit http://us.archive.ubuntu.com trusty Release.gpg                            
Get:4 http://dl.google.com stable Release [1,189 B]                            
Hit http://us.archive.ubuntu.com trusty-backports/main Sources                 
Ign http://extras.ubuntu.com trusty InRelease                                  
Get:5 http://security.ubuntu.com trusty-security InRelease [65.9 kB]           
Hit http://us.archive.ubuntu.com trusty-backports/restricted Sources           
Get:6 http://dl.google.com stable/main amd64 Packages [1,392 B]                
Hit http://ppa.launchpad.net trusty InRelease                                  
Hit http://us.archive.ubuntu.com trusty-backports/universe Sources             
Hit http://us.archive.ubuntu.com trusty-backports/multiverse Sources           
Hit https://repo.skype.com stable InRelease                                    
Get:7 http://extras.ubuntu.com trusty Release.gpg [72 B]                       
Hit http://us.archive.ubuntu.com trusty-backports/main amd64 Packages          
Hit https://repo.skype.com stable/main amd64 Packages                          
Hit http://us.archive.ubuntu.com trusty-backports/restricted amd64 Packages    
Hit http://us.archive.ubuntu.com trusty-backports/universe amd64 Packages      
Hit http://extras.ubuntu.com trusty Release                                    
Hit http://us.archive.ubuntu.com trusty-backports/multiverse amd64 Packages    
Hit http://ppa.launchpad.net trusty InRelease                                  
Hit http://us.archive.ubuntu.com trusty-backports/main i386 Packages           
Hit http://us.archive.ubuntu.com trusty-backports/restricted i386 Packages     
Hit http://extras.ubuntu.com trusty/main Sources                               
[... 62 lines cut ...]
Get:24 http://security.ubuntu.com trusty-security/universe Translation-en [110 kB]
Ign http://ppa.launchpad.net trusty Release.gpg                                
Hit http://ppa.launchpad.net trusty/main amd64 Packages
Hit http://ppa.launchpad.net trusty/main i386 Packages                  
Hit http://ppa.launchpad.net trusty/main Translation-en                 
Hit http://ppa.launchpad.net trusty/main amd64 Packages                       
Hit http://ppa.launchpad.net trusty/main i386 Packages                
Hit http://ppa.launchpad.net trusty/main Translation-en               
Ign http://ppa.launchpad.net trusty Release                                 
Err http://ppa.launchpad.net trusty/main amd64 Packages                        
  404  Not Found
Err http://ppa.launchpad.net trusty/main i386 Packages                         
  404  Not Found
Ign http://ppa.launchpad.net trusty/main Translation-en_US                     
Ign http://ppa.launchpad.net trusty/main Translation-en                        
Fetched 2,503 kB in 9s (253 kB/s)                                              
W: Failed to fetch http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu/dists/trusty/main/binary-amd64/Packages  404  Not Found

W: Failed to fetch http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu/dists/trusty/main/binary-i386/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.

```

In conclusion, really don't know what to do now.....

xlz, 2017-10-25:
> Then I try to install GLFW3 again by the comand sudo apt-add-repository ppa:floe/beignet

You were making some mistakes following README. The commands you listed are for OpenCL. Also it appears you have skipped this step "(Ubuntu 14.04 only) Download upgrade deb files."

closed 2017-10-25 by xlz