OpenKinect/libfreenect2 #923: ./Protonect freezes on Ubuntu 16.04

puren, 2017-10-27:
Hello,

I installed libfreenect on Ubuntu 16.04. When I run ./Protonect, it freezes and I have to reboot the system. It works fine with -nodepth.

I list the output of the suggested commands to run.

1. `git log -1 --oneline`
a01c4a6 docs: Update Ubuntu 14.04 OpenGL instructions
2. `uname -a`
Linux puren-Z87X-D3H 4.4.0-97-generic #120-Ubuntu SMP Tue Sep 19 17:28:18 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
3. `lsusb -t`
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 5: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 4, If 0, Class=Hub, Driver=hub/1p, 5000M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 5, If 1, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 5, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 9: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 10: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 11: Dev 5, If 0, Class=Imaging, Driver=usbfs, 480M
    |__ Port 11: Dev 5, If 1, Class=Vendor Specific Class, Driver=usbfs, 480M
    |__ Port 11: Dev 5, If 2, Class=Vendor Specific Class, Driver=ipheth, 480M
    |__ Port 12: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M

4. `lspci -nn`
00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller [8086:0c01] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 04)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection I217-V [8086:153b] (rev 04)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 04)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d4)
00:1c.3 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev d4)
00:1f.0 ISA bridge [0601]: Intel Corporation Z87 Express LPC Controller [8086:8c44] (rev 04)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c02] (rev 04)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 04)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GK110 [GeForce GTX TITAN] [10de:1005] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation GK110 HDMI Audio [10de:0e1a] (rev a1)
03:00.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev 41)

xlz, 2017-10-28:
Does it freeze with `-noviewer`?

puren, 2017-10-30:
No, it works fine with -noviewer.

xlz, 2017-10-30:
I think it has something to do with your graphics setup. If Protonect -noviewer works libfreenect2 itself should work alright.

puren, 2017-10-30:
Thank you for your answer. Do you mean the packages installed for opengl for instance? Do you have any suggestions to understand the actual problem in the graphics setup?

puren, 2017-11-01:
Hi,

I reinstalled Ubuntu 16.04.
For libfreenect2, I only installed the non-optional packages from the instructions.
Protonect works with -noviewer, -nodepth and -norgb. However when it runs without these options, it freezes the desktop and I have to reboot the computer. Do you have any idea why it runs depth and rgb individually without any problem but freezes when any option is selected? 

I am trying to understand the exact problem before starting to install new things. You said something about the graphic setup. I use Nouveau display driver. Can this cause the problem?

puren, 2017-11-01:
Selecting nvidia-340 in Additional Drivers solved the problem.

closed 2017-11-01 by puren