OpenKinect/libfreenect2 #924: Point cloud origin in the physical Kinect

AiSard, 2017-11-01:
Where is the point cloud origin in relation to an exterior feature of the _physical_ Kinect(v2)?
I understand that it is the center of the IR Sensor (?) but not where that sensor is in the Kinect.

Haven't been able to find it in any documentation, and most answers online seem to ignore the physical Kinect itself..

At wits end, if you can provide any help I'd be very thankful.

floe, 2017-11-01:
Good question - see the first image here: https://www.ifixit.com/Teardown/Xbox+One+Kinect+Teardown/19725#s55255

The IR sensor is the greenish lens, left-of-center. So it's located maybe 7-8 cm inwards from the left side of the casing.

xlz, 2017-12-01:
Answered.

closed 2017-12-01 by xlz