OpenKinect/libfreenect2 #925: Compiling C project which include libusb library - Ubuntu Terminal

CodeFreaklLinux, 2017-11-01:
Dear friends,
Hope everything be at your convenience.
I'm beginner in programing. I've installed libusb-1.0 ;
```
apt-get update
apt-get upgrade
sudo apt-get install libusb-1.0-0-dev
```
and dpkg -l libusb* shows;
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                     Version           Architecture      Description
+++-========================-=================-=================-=====================================================
ii  libusb-0.1-4:amd64       2:0.1.12-28       amd64             userspace USB programming library
ii  libusb-1.0-0:amd64       2:1.0.20-1        amd64             userspace USB programming library
ii  libusb-1.0-0-dbg:amd64   2:1.0.20-1        amd64             userspace USB programming library development files
ii  libusb-1.0-0-dev:amd64   2:1.0.20-1        amd64             userspace USB programming library development files
ii  libusb-1.0-doc           2:1.0.20-1        all               documentation for userspace USB programming
un  libusb-dev               <none>            <none>            (no description available)
un  libusbmuxd-tools         <none>            <none>            (no description available)
ii  libusbmuxd4:amd64        1.0.10-2ubuntu0.1 amd64             USB multiplexor daemon for iPhone and iPod Touch devi
```
after compiling my program;
`sudo -g -o genericdata generic.c -lusb-1.0`
following note will appear as I try to debug my program;
```
libusb_init (context=0x0) at ../../libusb/core.c:1983
1983	../../libusb/core.c: No such file or directory.
(gdb) quit
```
Please let me know what should I do.
Thanks

floe, 2017-11-01:
This just means that you don't have the source code for libusb-1.0 installed so you can't view the exact source line while debugging. Note that this only applies to the internal code of libusb, not to your program itself. BTW, is this in any way actually related to libfreenect2?

CodeFreaklLinux, 2017-11-01:
Thanks dear Floe for your reply
I'm beginner and I don't know libfreenect2, should I look for?
Is there any way to add "core.c" and its Essentials?

CodeFreaklLinux, 2017-11-01:
Dear Floe Florian Echtler, thanks for your reply
I just found;
https://github.com/OpenKinect/libfreenect2/issues/627
as it was closed I tried to make a new topic.

floe, 2017-11-01:
This support forum is for libfreenect2 which is built on top of libusb-1.0. You should rather ask your questions here: https://github.com/libusb/libusb/issues

closed 2017-11-01 by floe