OpenKinect/libfreenect2 #927: error C2001:

r-fujikura, 2017-11-02:
Overview Description:
Hi, everyone.
I'm a college student in Japan.
I would like to implement elasticfusion, but I guess I have to use libfreenect2 to implement with Kinect v2.
So I have a  error.
I want you to tell me how to solve.

Version, Platform, and Hardware Bug Found:
Windows 10, cmake(3.10.0-rc4), visual studio 2015 community,kinect v2  

I ran libusb_2015.sln, but this kind of error occurred in visual studio 2015.
I am in trouble, please help me.

Actual Results:
1>------ ビルド開始: プロジェクト:libusb-1.0 (static), 構成:Debug x64 ------
2>------ ビルド開始: プロジェクト:libusb-1.0 (dll), 構成:Debug x64 ------
3>------ ビルド開始: プロジェクト:libusb-usbdk-1.0 (dll), 構成:Debug x64 ------
4>------ ビルド開始: プロジェクト:libusb-usbdk-1.0 (static), 構成:Debug x64 ------
1>  strerror.c
1>..\libusb\strerror.c : warning C4819: ファイルは、現在のコード ページ (932) で表示できない文字を含んでいます。データの損失を防ぐために、ファイルを Unicode 形式で保存してください。
2>  strerror.c
2>..\libusb\strerror.c : warning C4819: ファイルは、現在のコード ページ (932) で表示できない文字を含んでいます。データの損失を防ぐために、ファイルを Unicode 形式で保存してください。
4>  strerror.c
4>..\libusb\strerror.c : warning C4819: ファイルは、現在のコード ページ (932) で表示できない文字を含んでいます。データの損失を防ぐために、ファイルを Unicode 形式で保存してください。
3>  strerror.c
3>..\libusb\strerror.c : warning C4819: ファイルは、現在のコード ページ (932) で表示できない文字を含んでいます。データの損失を防ぐために、ファイルを Unicode 形式で保存してください。
1>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(426): warning C4324: 'libusb_device': アラインメント指定子のために構造体がパッドされました
1>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(438): warning C4324: 'libusb_device_handle': アラインメント指定子のために構造体がパッドされました
1>..\libusb\strerror.c(111): error C2001: 定数が 2 行目に続いています。
1>..\libusb\strerror.c(117): error C2001: 定数が 2 行目に続いています。
2>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(426): warning C4324: 'libusb_device': アラインメント指定子のために構造体がパッドされました
2>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(438): warning C4324: 'libusb_device_handle': アラインメント指定子のために構造体がパッドされました
2>..\libusb\strerror.c(111): error C2001: 定数が 2 行目に続いています。
2>..\libusb\strerror.c(117): error C2001: 定数が 2 行目に続いています。
4>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(426): warning C4324: 'libusb_device': アラインメント指定子のために構造体がパッドされました
4>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(438): warning C4324: 'libusb_device_handle': アラインメント指定子のために構造体がパッドされました
4>..\libusb\strerror.c(111): error C2001: 定数が 2 行目に続いています。
4>..\libusb\strerror.c(117): error C2001: 定数が 2 行目に続いています。
3>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(426): warning C4324: 'libusb_device': アラインメント指定子のために構造体がパッドされました
3>c:\users\t-takahashi\desktop\libfreenect2-master\libfreenect2-master\depends\libusb_src\libusb\libusbi.h(438): warning C4324: 'libusb_device_handle': アラインメント指定子のために構造体がパッドされました
3>..\libusb\strerror.c(111): error C2001: 定数が 2 行目に続いています。
3>..\libusb\strerror.c(117): error C2001: 定数が 2 行目に続いています。
========== ビルド: 0 正常終了、4 失敗、6 更新不要、0 スキップ ==========
Expected Results:

Reproducibility:

Additional Information:

floe, 2017-11-02:
Since very few people here can read Japanese, we won't be able to help you unless you can post an English error message. But given that both warning C4819 and error C2001 are related to encoding issues, you might first look if your Visual Studio installation is properly configured for UTF8-encoded source files.

closed 2017-11-02 by floe

r-fujikura, 2017-11-03:
Thank you for your patience. How can I set visual studio for UTF - 8 encoded source files you are talking about? I appreciate it if you let me know.

floe, 2017-11-03:
I don't use Visual Studio myself, so I can't tell you. I found the information above using Google, so you should try that, too.

r-fujikura, 2017-11-03:
I was able to solve this problem. Thank you very much. By the way, do you know elasticfusion in relation?

floe, 2017-11-03:
Just by name - never used it myself.

r-fujikura, 2017-11-08:
OK. Thank you.