OpenKinect/libfreenect2 #928: Could not Import freenect2

sowrabhnayak, 2017-11-06:
I have followed the steps for the installation of libfreenect2 as given in this link ( https://github.com/OpenKinect/libfree... Ubuntu 14.04.1) i have installed it at /home/user_name/ 
I have to import freenect2 just like this https://rjw57.github.io/freenect2-python/
But i'm not able to import freeenct2 and it gives me this error:

import freenect2 Traceback (most recent call last): File "<stdin>", line 1, in <module> ImportError: No module named freenect2

Or do i have any other option....

totovr, 2017-11-11:
I think that is better to follow the original tutorial 

https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux

ArmSorawis, 2018-07-05:
Have you already find way to solve this problem? If you knew just tell me please .

totovr, 2018-07-05:
Follow the original tutorial and it will work

closed 2018-11-18 by xlz