OpenKinect/libfreenect2 #931: libusb problem for vcpkg integration

jasjuang, 2017-11-09:
I am trying to integrate libfreenect2 into [vcpkg](https://github.com/Microsoft/vcpkg). One problem I ran into is the dependency with libusb. There is already a libusb port in vcpkg, however according to https://github.com/OpenKinect/libfreenect2/blob/master/depends/install_libusb_vs2015.cmd, the change applied at the fork https://github.com/xlz/libusb has to be applied on top of https://github.com/libusb/libusb in order to get libfreenect2 to work. My question is why not simply submit a PR to merge the xlz/libusb into libusb/libusb? I would like to know since applying this patch in vcpkg is not trivial.

xlz, 2017-11-09:
The support for USB3 isochronous transfer is not mature in libusb. libusb provides it through the UsbDk backend which is not mature. The fork ships a patch for the libusbK backend which was well-tested. This patch was submitted to libusb years ago but they had their own plans about how the feature was supposed to be implemented and the actual implementation was slow as glaciers. Take a look at all referenced links in https://github.com/libusb/libusb/issues/46.

jasjuang, 2017-11-09:
@xlz Thanks for the clarification. I will try to apply your libusb patch onto the vcpkg libusb port then.

closed 2017-11-09 by jasjuang

xlz, 2017-11-09:
If you are fine with using UsbDk without libusbk you can just use the stock libusb. The build system of libfreenect2 can handle it without libusbK.

jasjuang, 2017-11-09:
For some reason UsbDk doesn't work for me. I am only able to see the kinect working through the libusbK alternative.

jasjuang, 2017-11-21:
@xlz Just so you know that the vcpkg port got merged recently, so installing libfreenect2 on Windows would be as simple as `vcpkg install libfreenect2`