OpenKinect/libfreenect2 #935: [Debug] [DepthPacketStreamParser] image data too short!

ngageorange, 2017-11-24:
Hello,
Having a very difficult time trying to get Kinectv2 to work on my fresh Ubuntu 16.04 machine. A colleague of mine has the exact same hardware which runs this fine so I can only imagine it's a very odd issue...
Initially thought because I was on ASMedia controller, but my colleague is as well and his works fine. Nevertheless, I tried another controller on a PCI x8 bus..

The error on running `Protonect` is:
`[Debug] [DepthPacketStreamParser] image data too short!`

The window is frozen without an image and the console is flooded with the output above.

`dmesg` gives the following output:
`
.....
.....
[  105.145730] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.145731] xhci_hcd 0000:0a:00.0: URB req 0 and actual 28312 transfer length mismatch
[  105.146311] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.146313] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.147242] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.147245] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.147248] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.147249] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.147251] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.147252] xhci_hcd 0000:0a:00.0: URB req 0 and actual 33792 transfer length mismatch
[  105.147255] xhci_hcd 0000:0a:00.0: URB req 0 and actual 28312 transfer length mismatch
`

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
`a01c4a6 docs: Update Ubuntu 14.04 OpenGL instructions`
2. `uname -a`
`Linux waffle 4.13.0-17-generic #20~16.04.1-Ubuntu SMP Mon Nov 6 14:18:00 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`
3. `lsusb -t`
`
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 14: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 480M
    |__ Port 14: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 480M
`
4. `lspci -nn`
`
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1450]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:1451]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1452]
00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:1453]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1452]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1452]
00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:1453]
00:03.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:1453]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1452]
00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1452]
00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:1454]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1452]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:1454]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 59)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1460]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1461]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1462]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1463]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1464]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1465]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1466]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1467]
01:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b9] (rev 02)
01:00.1 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b5] (rev 02)
01:00.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b0] (rev 02)
02:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b4] (rev 02)
02:02.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b4] (rev 02)
02:03.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b4] (rev 02)
02:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b4] (rev 02)
02:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b4] (rev 02)
02:07.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b4] (rev 02)
06:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:1343]
07:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
08:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
09:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1c03] (rev a1)
09:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10f1] (rev a1)
0a:00.0 USB controller [0c03]: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller [1912:0015] (rev 02)
0b:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Device [1022:145a]
0b:00.2 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Device [1022:1456]
0b:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:145c]
0c:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Device [1022:1455]
0c:00.2 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] [1022:7901] (rev 51)
0c:00.3 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Device [1022:1457]
`

Any help would be gratefully appreciated.

-Ahmed

xlz, 2017-11-24:
```
01:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:43b9] (rev 02)
06:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:1343]
0a:00.0 USB controller [0c03]: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller [1912:0015] (rev 02)
0b:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:145c]
```
It seems you've got quite a few USB controllers. The AMD ones seem quite new (hence flaky support). But the Renesas uPD720202 one is old and should work. Do you have multiple USB ports? Maybe try different ports and see if anyone is connected to uPD720202 and works?

ngageorange, 2017-11-27:
Hi xlz,
Many thanks for your quick response and apologies for my late reply!

It turns out the newer kernel, 4.13.0-17 has a regression in it that stops libfreenect from working as per the above errors...4.13.0-16 works a treat on either the AMD and the Renesas USB buses.

For the sake of completion, the latest BIOS also needs to be updated as well as the latest microcode from AMD to work. Upgrading to 4.13.0-17 even with the latest microcode and bios show several USB related errors.

Any thoughts on how we can feed this back upstream?

-Ahmed

xlz, 2017-11-27:
I recommend you report a bug to Ubuntu with detailed description for reproducing this bug and detailed logs.

If you care enough you can bisect the specific commit that cause the regression, which would make your bug report look better for upstream. Bisecting requires setting up the kernel build environment which may take some time though.

davclark, 2017-12-04:
FWIW - this just bit me on an Intel NUC (7th gen i5), which obviously has intel USB drivesrs, and reverting the kernel version restored functionality. So, it seems like a pretty general issue for Ubuntu at least, if not for the kernel itself.

xlz, 2017-12-05:
It's possible to debug this (I won't be able to though so I'm just writing theories here).

The error message `URB req 0 and actual 28312 transfer length mismatch` is from here
https://github.com/torvalds/linux/blob/894025f24bd028942da3e602b87d9f7223109b14/drivers/usb/host/xhci-ring.c#L1897
```c
	if (urb->actual_length > urb->transfer_buffer_length) {
		xhci_warn(xhci, "URB req %u and actual %u transfer length mismatch\n",
			  urb->transfer_buffer_length, urb->actual_length);
```
`transfer_buffer_length` being 0 looks wrong. `transfer_buffer_length` is assigned here
https://github.com/torvalds/linux/blob/894025f24bd028942da3e602b87d9f7223109b14/drivers/usb/core/devio.c#L1688
```c
	as->urb->transfer_buffer_length = uurb->buffer_length;
```
`uurb->buffer_length` is assigned here
https://github.com/torvalds/linux/blob/894025f24bd028942da3e602b87d9f7223109b14/drivers/usb/core/devio.c#L1566
```c
		for (totlen = u = 0; u < number_of_packets; u++) {
[...]
			totlen += isopkt[u].length;
		}
		u *= sizeof(struct usb_iso_packet_descriptor);
		uurb->buffer_length = totlen;
```
So for `transfer_buffer_length` to be 0, all `isopkt[u].length` is 0. `isopkt` is from userspace, the `iso_frame_desc` field of struct `usbdevfs_urb`, which is passed via `USBDEVFS_SUBMITURB` ioctl.

I would follow `USBDEVFS_SUBMITURB` into libusb but libusb already has got this bug reported: https://github.com/libusb/libusb/pull/365.

According to their discussion this is actually a real kernel regression which breaks userspace. So there is nothing to fix in userspace.

Introduced in https://github.com/torvalds/linux/commit/fa1ed74eb1c233be6131ec92df21ab46499a15b6 (mainline 4.14-rc4, linux-stable 4.13.6, ubuntu 4.13.0-17.20\~16.04.1). 
Fixed in https://github.com/torvalds/linux/commit/845d584f41eac3475c21e4a7d5e88d0f6e410cf7 (mainline 4.14-rc6, linux-stable 4.13.10, ubuntu 4.13.0-18.21~16.04.1). (The offending patch was killed in the mainline but slipped through the cracks into stable!)

Read the commit description for detail.

Ubuntu users can resolve this issue by updating to 4.13.0-18.21~16.04.1 (in proposed repo at the time of writing).

closed 2017-12-05 by xlz

davclark, 2017-12-05:
I'm humbled! Thank you for the effort in making sense of this issue.