OpenKinect/libfreenect2 #939: Application Error 0x000007b when running Protonect.exe after libfreenect2 sucessful build

pauloguerraf, 2017-11-30:
Hi, 

First of all thank you for your work on this library. I have successfully built libfreenect2 from source :
[image]
[image]

But unfortunately when I try to run Protonect.exe I get the following error:
[image]

Any idea why is this could be happening?
I have tried your libfreenect2-0.2.0-libusbk-vs2015-x64 compiled release and it works correctly. Is there something I am missing when trying to build my version from scratch?

I am using the source code downloaded from your github master branch, running Windows 10

Thank you in advance for your help.

Best,

Paulo

xlz, 2017-12-01:
You built from source using this script `make_release_msvc.cmd`. It is not documented and is used by the developers to make release, which means I can't answer why you had this problem. The error 0xc000007b means missing DLLs. Maybe you can figure it out yourself.

The documentation for building from source is in README.

closed 2017-12-01 by xlz

pauloguerraf, 2017-12-01:
I get the exact same error when using 
cmake .. -G "Visual Studio 14 2015 Win64"
and 
cmake --build . --config RelWithDebInfo --target install

[image]
[image]

Thank you