OpenKinect/libfreenect2 #940: Plugged Kinect2 locks USB devices

digitalmeltingculture, 2017-12-04:
Overview Description:

Hi all,
I installed libfreenect2 on Ubuntu Studio 16.04 and 17.04 and tested it successfully  with some applications.
Then, i developed some code based on libfreenect2 that sends output to an external audiobox
For instance, i use the Presonus Audiobox VSL 1818 VSL. The kinect one and the audiobox work well separately, but if i launch the application with both devices plugged on, I can't hear any audio. Then, if i plug the kinect off, the audio plays correctly.

I wrote this as issue here, because also a similiar issue happened to me, The same application works in such a way that the PC executes boot and shutdown at prefixed times. But when I restart the PC the usb ports seem locked. I need to plug the kinect off and reboot the PC to make they work.

Any ideas for these scenarios?

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2017-12-05:
Kinect includes some audio devices. You should check your audio system and see if it is configured to use unexpected audio devices.

digitalmeltingculture, 2017-12-17:
Thanks @xlz. Do you think it's a question of supported audio devices? If so, is there a supported devices list?
I wrote this issue because, as i mentioned, not only my audio device but also other usb devices, such as hub, mouse and keyboard are often excluded when the Kinect is plugged in.
I need to plug them out, restart the PC, plug them in and hope they work.

Did this already happened to anyone?

closed 2018-11-18 by xlz