OpenKinect/libfreenect2 #944: can't build or execute on Windows 7

normanhantke, 2017-12-08:
**Overview:**

Two different problems: I can't build master branch and latest release doesn't work.

Windows 7 64bit
cmake version 3.7.0-rc1
visual studio 15
kinect v2 model 1656

I have everything running just fine on linux, so the hardware should be okay.

**Build problem:**

TURBOJPEG_WORKS test fails although he does find the lib and the headers.

output:
D:\Windows_projects\libfreenect2\build>cmake .. -G "Visual Studio 14 2015 Win64"
-- The C compiler identification is MSVC 19.0.23918.0
-- The CXX compiler identification is MSVC 19.0.23918.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE)
-- Performing Test LIBUSB_WITH_LIBUSBK
-- Performing Test LIBUSB_WITH_LIBUSBK - Success
-- Performing Test LIBUSB_WITH_USBDK
-- Performing Test LIBUSB_WITH_USBDK - Failed
CMake Warning at cmake_modules/FindLibUSB.cmake:55 (MESSAGE):
  UsbDk device driver is not found.  Fall back to libusbK.
Call Stack (most recent call first):
  CMakeLists.txt:93 (FIND_PACKAGE)

-- Found LibUSB: D:/Windows_projects/libfreenect2/depends/libusb/MS64/dll/libusb-1.0.lib
-- Found JPEG: C:/Users/hantke/AppData/Local/Continuum/Anaconda3/Library/lib/jpeg.lib
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS)
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Failed
CMake Error at C:/Program Files/CMake/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find TurboJPEG (missing: TURBOJPEG_WORKS)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:240 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "D:/Windows_projects/libfreenect2/build/CMakeFiles/CMakeOutput.log".
See also "D:/Windows_projects/libfreenect2/build/CMakeFiles/CMakeError.log".

**Protonect.exe problem:**

_with usbdk_1.0.19:_
device manager -> Xbox NUI Sensor -> Properties says "The drivers for this device are not installed. (Code 28)"
Is this normal or does it indicate a problem with the usbdk installation?

Output Protonect Release 0.2.0 (usbdk variant):
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\bin\Protonect.exe [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 5 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @10:1 with serial 501823241942
[Info] [Freenect2Impl] found 1 devices
libusb: error [usbdk_open] Redirector startup failed
[Error] [Freenect2Impl] failed to open Kinect v2: @10:1 LIBUSB_ERROR_OTHER Other error
failure opening device!

_with libusbk 3.0.7.0:_
device manager shows everything ok

Output Protonect Release 0.2.0 (libusbk variant):
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\bin\Protonect.exe [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_02D8#501823241942': [87] The parameter is incorrect.
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_02D9#6&4D260FB&0&1': [87] The parameter is incorrect.
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_02D9#6&4D260FB&0&3': [87] The parameter is incorrect.
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

xlz, 2017-12-12:
> TURBOJPEG_WORKS test fails although he does find the lib and the headers.

How did you manage to build it?

> usbdk_1.0.19:
device manager -> Xbox NUI Sensor -> Properties says "The drivers for this device are not installed. (Code 28)"
> libusb: error [windows_get_device_list] program assertion failed - existing device should share parent

Is the device installed correctly or something. You can try installing the official Microsoft SDK and see if that works.

normanhantke, 2017-12-13:
The SDK does not support Windows 7...

The build problem is solved: I accidentally had a 32bit version of turbojpeg installed.
Running Protonect after building from source gives me another output (with usbdk):

Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 5 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @10:1 with serial 501823241942
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to get max iso packet size! LIBUSB_ERROR_NOT_FOUND Entity not found. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @10:1
failure opening device!
libusb: warning [libusb_exit] some libusb_devices were leaked

I guess my only option is to switch to Windows 8/10?

xlz, 2017-12-13:
Try libusbk?

normanhantke, 2017-12-14:
Built from source; with libusbk:

D:\Windows_projects\libfreenect2\build>.\install\bin\Protonect.exe
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_02D8#501823241942': [87] The parameter is incorrect.
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_02D9#6&4D260FB&0&1': [87] The parameter is incorrect.
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_02D9#6&4D260FB&0&3': [87] The parameter is incorrect.
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

AlexMorgand, 2018-01-04:
Having the same issue. Did you found a workaround?

I had the issue
`[Error] [protocol::UsbControl] failed to get max iso packet size! LIBUSB_ERROR_NOT_FOUND Entity not found. Try debugging with environment variable: export LIBUSB_DEBUG=3 .`

`$  ./install/bin/Protonect.exe
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\am237982\Desktop\cea\CEA\Alexandre\Dev\Kinect2\build\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 5 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @59:2 with serial 201458134747
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Info] [Freenect2Impl] device unavailable right now, retrying
[Error] [Freenect2Impl] failed to open Kinect v2: @59:2 LIBUSB_ERROR_OTHER Other error
failure opening device!`

I'm using UsbDk.

xlz, 2018-01-04:
Does thie prebuilt binaries work? https://github.com/OpenKinect/libfreenect2/releases/tag/v0.2.0

AlexMorgand, 2018-01-04:
Got this.

> $ ./Protonect.exe
> Version: 0.2.0
> Environment variables: LOGFILE=<protonect.log>
> Usage: C:\Users\am237982\Downloads\libfreenect2-0.2.0-usbdk-vs2013-x64\libfreenect2-0.2.0-usbdk-vs2013-x64\bin\Protonect.exe [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
>         [-noviewer] [-norgb | -nodepth] [-help] [-version]
>         [-frames <number of frames to process>]
> To pause and unpause: pkill -USR1 Protonect
> [Info] [Freenect2Impl] enumerating devices...
> [Info] [Freenect2Impl] 5 usb devices connected
> [Info] [Freenect2Impl] found valid Kinect v2 @110:2 with serial 201458134747
> [Info] [Freenect2Impl] found 1 devices
> [Error] [Freenect2Impl] failed to open Kinect v2: @110:2 LIBUSB_ERROR_OTHER Other error

AlexMorgand, 2018-01-04:
And just got back to the previously mentioned error:

> $ ./Protonect.exe
> Version: 0.2.0
> Environment variables: LOGFILE=<protonect.log>
> Usage: C:\Users\am237982\Desktop\cea\CEA\Alexandre\Dev\Kinect2\build\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
>         [-noviewer] [-norgb | -nodepth] [-help] [-version]
>         [-frames <number of frames to process>]
> To pause and unpause: pkill -USR1 Protonect
> [Info] [Freenect2Impl] enumerating devices...
> [Info] [Freenect2Impl] 5 usb devices connected
> [Info] [Freenect2Impl] found valid Kinect v2 @125:2 with serial 201458134747
> [Info] [Freenect2Impl] found 1 devices
> [Info] [Freenect2DeviceImpl] opening...
> [Error] [protocol::UsbControl] failed to get max iso packet size! LIBUSB_ERROR_NOT_FOUND Entity not found. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
> [Info] [Freenect2DeviceImpl] closing...
> [Info] [Freenect2DeviceImpl] releasing usb interfaces...
> [Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
> [Info] [Freenect2DeviceImpl] closing usb device...
> [Info] [Freenect2DeviceImpl] closed
> [Error] [Freenect2Impl] failed to open Kinect v2: @125:2
> failure opening device!
> libusb: warning [libusb_exit] some libusb_devices were leaked

`export LIBUSB_DEBUG=3 .` doesn't seem to do anything for me. Maybe I could try to debug with VS directly.

AlexMorgand, 2018-01-04:
Fixed it by using libusbk instead.  Protonect works like a charm.

AlexMorgand, 2018-01-04:
Interesting founding, if you don't want to uninstall UsbDk (can broke the drivers for the keyboard and mouse), you can just comment line 48 of FindLibUSB.cmake.

>  #CHECK_C_SOURCE_RUNS("#include <windows.h>\nint main(){return !LoadLibraryA(\"UsbDkHelper\");}" LIBUSB_WITH_USBDK)
to just use libusbk.

I have a working version with libusbk working VS 2013 and I tried the pre-built binaries as well.

xlz, 2018-01-05:
LIBUSB_USE_USBDK is a turnable flag in CMake. You should use Cmake to turn it off. Not necessary to edit the sourc.e

closed 2018-11-18 by xlz