OpenKinect/libfreenect2 #946: GLFW on Win 8.1  - not initialized error

margokhokhlova, 2017-12-12:
I am trying to compile the project on Win 8.1. All the dependencies are installed *cuda 7.5, glfw 3.1, openc cl, turbo jpeg), GLFW examples work, but I have the same error all the time:

65537 The GLFW library is not initialized
Failed to create opengl window.
It appears after the device initialization and start of the acquisition.

The test of freenect-openni is OK, but displays the data from one kinect only when multiple kinects are attached.
Protonect works in -noviewer mode, detects both cameras.

Please, if someone have any idea what am I doing wrong, let me know.

SirDifferential, 2018-02-26:
I saw this on Windows 10. This problem appears to be caused by the latest version of GLFW not liking `glfwInit()` being called multiple times due to a Windows windowing function call causing issues. I didn't investigate this further yet, but you can solve this temporarily by making sure `glfwInit()` is only called once.

* With OpenCL viewer, glfwInit() is called inhttps://github.com/OpenKinect/libfreenect2/blob/master/src/logging.cpp#L202
* With OpenGL, it's called at https://github.com/OpenKinect/libfreenect2/blob/master/src/opengl_depth_packet_processor.cpp#L862
* Protonect calls it here https://github.com/OpenKinect/libfreenect2/blob/master/examples/viewer.cpp#L19

mark-berobot, 2020-01-18:
Just ran into this on Windows 10, compiling only with OpenGL support. It does run with `-nowindow` which seems to confirm @SirDifferential 's theory. However it would point to a bug(?) in GLFW, because the docs https://www.glfw.org/docs/latest/intro_guide.html#intro_init say ` The library only needs to be initialized once and additional calls to an already initialized library will return GLFW_TRUE immediately.`

mark-berobot, 2020-01-18:
`glfwInit()` is also called in `logging.cpp`. Maybe that code path should be removed.

mark-berobot, 2020-01-19:
I went into `config.h` and commented out `#define LIBFREENECT2_WITH_OPENGL_SUPPORT` and enabled `#define LIBFREENECT2_WITH_CXX11_SUPPORT ` and recompiled. Then `Protonect cl` works without errors.