OpenKinect/libfreenect2 #947: Libusb errors windows 10

fred-dev, 2017-12-18:
Overview Description:

I am guessing this is not a bug but my own fault. Is there a forum or wiki that is a more appropriate place to ask questions (I dont find one). I did go through the FAQ and troubleshooting, but am getting this error. Any light would be great and if there is a more appropiate place to ask questions please let me know.

`[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00003234] libusb: debug [libusb_init] created default context
[ 0.000000] [00003234] libusb: debug [libusb_init] libusb v1.0.19.10905
[ 0.001003] [00003234] libusb: debug [windows_init] Windows 8 64-bit
[ 0.001003] [00003234] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.002006] [00003234] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.002006] [00003234] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.002006] [00003234] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.002006] [00003234] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.003009] [00003234] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.004012] [00003e8c] libusb: debug [windows_clock_gettime_threaded] hires timer available (Frequency: 4101557 Hz)
[ 0.004012] [00003234] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004012] [00003234] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.004012] [00003234] libusb: debug [usbi_add_pollfd] add fd 1 events 1
[ 0.004012] [00003d20] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.005015] [00003234] libusb: debug [libusb_get_device_list]
[ 0.005015] [00003d20] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.005015] [00003d20] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[ 0.005015] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [2FC]
[ 0.005015] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [3ED]
[ 0.005015] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [C9]
[ 0.005015] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.005015] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005015] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [271]
[ 0.005015] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.005015] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005015] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [167]
[ 0.005015] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.005015] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005015] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [3A2]
[ 0.006017] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.006017] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.006017] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [1E5]
[ 0.006017] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.006017] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.006017] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [211]
[ 0.007020] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.007020] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.007020] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [26F]
[ 0.007020] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.007020] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.007020] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [E6]
[ 0.008023] [00003234] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.008023] [00003234] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.008023] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [25B]
[ 0.009025] [00003234] libusb: debug [windows_get_device_list] found existing device for session [1E5] (0.0)
[ 0.010027] [00003234] libusb: debug [init_device] (bus: 2, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&1867B3C4&0&0'
[ 0.010027] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [2C5]
[ 0.010027] [00003234] libusb: debug [init_device] got bus number from ancestor #3
[ 0.010027] [00003234] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.010027] [00003234] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 62 bytes)
[ 0.011030] [00003234] libusb: debug [init_device] (bus: 2, addr: 8, depth: 1, port: 3): '\\.\USB#VID_2A70&PID_F003&MI_00#7&2811DBA9&0&0000'
[ 0.011030] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [13]
[ 0.011030] [00003234] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.011030] [00003234] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 62 bytes)
[ 0.011030] [00003234] libusb: debug [init_device] (bus: 2, addr: 8, depth: 1, port: 3): '\\.\USB#VID_2A70&PID_F003#43A942B6'
[ 0.011030] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [161]
[ 0.011030] [00003234] libusb: debug [init_device] got bus number from ancestor #2
[ 0.012032] [00003234] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.012032] [00003234] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.012032] [00003234] libusb: debug [init_device] (bus: 2, addr: 13, depth: 1, port: 6): '\\.\USB#VID_28DE&PID_2101#F108CFF2C2'
[ 0.012032] [00003234] libusb: debug [windows_get_device_list] allocating new device for session [2E1]
[ 0.012032] [00003234] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.012032] [00003234] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.012032] [00003234] libusb: debug [init_device] (bus: 2, addr: 4, depth: 1, port: 3): '\\.\USB#VID_045E&PID_07F8&MI_00#6&247964C7&0&0000'
[... 116 lines cut ...]
[ 0.027072] [00003234] libusb: debug [get_api_type] driver(s): libusbK
[ 0.027072] [00003234] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.028075] [00003234] libusb: debug [windows_get_device_list] found existing device for session [36B] (1.4)
[ 0.028075] [00003234] libusb: debug [windows_get_device_list] setting HID interface for [161]:
[ 0.028075] [00003234] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_28DE&PID_2101#8&254E3913&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.028075] [00003234] libusb: debug [windows_get_device_list] setting HID interface for [10]:
[ 0.028075] [00003234] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_045E&PID_00CB#6&1EB9C568&1&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.028075] [00003234] libusb: debug [htab_hash] hash collision ('\\.\USB#VID_28DE&PID_2101#48F8A769E5' vs '\\.\PCI#VEN_8086&DEV_A2AF&SUBSYS_50071458&REV_00#3&11583659&1&A0')
[ 0.028075] [00003234] libusb: debug [windows_get_device_list] setting HID interface for [3FC]:
[ 0.028075] [00003234] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_28DE&PID_2101#8&3A91E5E5&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.030080] [00003234] libusb: debug [windows_get_device_list] setting HID interface for [1E0]:
[ 0.030080] [00003234] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_0BB4&PID_2C87#8&144AEC52&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.030080] [00003234] libusb: debug [get_api_type] driver(s): libusbK
[ 0.030080] [00003234] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.030080] [00003234] libusb: debug [libusb_get_device_descriptor]
[ 0.030080] [00003234] libusb: debug [libusb_get_device_descriptor]
[ 0.031083] [00003234] libusb: debug [libusb_unref_device] destroy device 2.8
[ 0.031083] [00003234] libusb: debug [libusb_get_device_descriptor]
[ 0.031083] [00003234] libusb: debug [libusb_unref_device] destroy device 2.8
[ 0.031083] [00003234] libusb: debug [libusb_get_device_descriptor]
[ 0.031083] [00003234] libusb: debug [libusb_unref_device] destroy device 2.13
[ 0.031083] [00003234] libusb: debug [libusb_get_device_descriptor]
[ 0.031083] [00003234] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.031083] [00003234] libusb: debug [libusb_get_device_descriptor]
[ 0.032086] [00003234] libusb: debug [libusb_get_device_descriptor]`

xlz, 2017-12-19:
I had a good looking but it seems you didn't include the error part in the log.

fred-dev, 2017-12-19:
The error I get (without LIBUSB_DEBUG=4) is just

```
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] all submissions failed. Try debugging with environment variable: LIBUSB_DEBUG=4.
```

The above output is what I get when I add LIBUSB_DEBUG=4
I dont see an explicit error either.

xlz, 2017-12-19:
The default libusb doesn't work, will result in LIBUSB_ERROR_NOT_SUPPORTED. We use a patched libusb. You should have had this step `.\install_libusb_vs2013.cmd`. There is probably some error during building libusb, can you post the output of `.\install_libusb_vs2013.cmd`?

xlz, 2017-12-19:
You can try the prebuilt binaries https://github.com/OpenKinect/libfreenect2/releases/tag/v0.2.0 and see if it is the case.

fred-dev, 2017-12-19:
Perfect, it was an unpatched libsub.dll, it works great now. 

Thanks a lot for your help.

closed 2017-12-20 by xlz