OpenKinect/libfreenect2 #948: TODO for streaming, recording, and replay

smokhov, 2017-12-19:
This TODO list is being gathered from the following PRs #555, #602, #937, #942, where ideas, items, or issues were raised but were not in scope/on scale for those PRs and should be a subject for separate RP(s):

- @brendandburns: `Freenect2ReplayDevice`, does not support RGB yet (#555)
- @PyrApple: `Streamer` does not support RGB yet (#602)
- @PyrApple: `ProtonectSR` to build on Windows (#602)
- @floe: gstreamer (#602)
- @xlz: How to load p0 tables (Read `prefix_123_001.p0table?`) (#942)
- @xlz: How to replay: realtime by timestamp? There should be no loss but the present logic will have packet loss so more work is needed. (replace ready() to synchronous calls)  (#942)
    - @xlz: How do you synchronize RGB with depth replay orders, how do you handle cross-thread synchronous data flow, etc etc.
- @smokhov: I see you (@xlz) really made it match a lot more with regular device and its API by doing Freenect2ReplayImpl, etc. Given that do they (Freenect2 and Freenect2Replay) deserve now a common parent since they are so a like? I could use that parent in the other PR in the tool that uses the device; if not it is also OK, can if them out (#942)
    - @xlz: Sharing a common parent class - they could. But there isn't much duplication right now so extra abstraction isn't as necessary as it appears.

smokhov, 2017-12-20:
Testing... got it to record now re-sync'ing and trying to get the playback to work using the replay device instead of blender. Apparently, OpenCV does not like `.depth` extension and is all fussy about what kind of image types it can save. For now my minimal code off the merges is this:
```patch
diff --git a/src/libfreenect2.cpp b/src/libfreenect2.cpp
index baf76c9..eb04712 100644
--- a/src/libfreenect2.cpp
+++ b/src/libfreenect2.cpp
@@ -1476,12 +1476,13 @@ void Freenect2ReplayDevice::run()
       continue;
     }

-    if (hasSuffix(frame, ".depth"))
+    if (hasSuffix(frame, ".jpg"))
     {
       std::ifstream fd(frame.c_str());

       if(!fd)
       {
+        LOG_ERROR << "what: " << strerror(errno);
         LOG_ERROR << "failed to open replay frame: " << frame << ", skipping...";
         continue;
       }
@@ -1490,15 +1491,22 @@ void Freenect2ReplayDevice::run()
       size_t length = fd.tellg();
       fd.seekg(0, fd.beg);

-      if(length != buffer_size_)
+      LOG_DEBUG << "length: " << length;
+
+      if(length > buffer_size_)
       {
         LOG_ERROR << "file length: " << length
-                  << "exceeds depth image buffer size: "
+                  << " exceeds depth image buffer size: "
                   << buffer_size_ << "; skipping...";
         continue;
       }

+      LOG_DEBUG << "buffer_seize_: " << buffer_size_;
+
       fd.read(reinterpret_cast<char*>(packet_.memory->data), length);
+
+      LOG_DEBUG << "read: " << fd.gcount();
+
       if(!fd || (size_t)fd.gcount() != length)
       {
         LOG_ERROR << "failed to read replay frame: " << frame << ": "
@@ -1506,6 +1514,8 @@ void Freenect2ReplayDevice::run()
         continue;
       }

+      LOG_DEBUG << "pipeline ready? " << pipeline_->getDepthPacketProcessor()->ready();
+
       if(pipeline_->getDepthPacketProcessor()->ready())
       {
         packet_.timestamp = timestamp_sequence[0];
@@ -1514,7 +1524,9 @@ void Freenect2ReplayDevice::run()
         packet_.buffer_length = length;

         pipeline_->getDepthPacketProcessor()->process(packet_);
+      LOG_DEBUG << "replay packet processed";
         pipeline_->getDepthPacketProcessor()->allocateBuffer(packet_, buffer_size_);
+      LOG_DEBUG << "replay packet new buffer allocated";
       }
       else
       {
@@ -1522,6 +1534,8 @@ void Freenect2ReplayDevice::run()
           << "skipping a replay depth packet for " << frame
           << " as depth processor is not ready";
       }
+
+      LOG_DEBUG << "after pipeline ";
     }
   }
 }
diff --git a/tools/streamer_recorder/ProtonectSR.cpp b/tools/streamer_recorder/ProtonectSR.cpp
index 9dd7eb4..771ed66 100644
--- a/tools/streamer_recorder/ProtonectSR.cpp
+++ b/tools/streamer_recorder/ProtonectSR.cpp
@@ -168,8 +168,7 @@ int main(int argc, char *argv[])

 /// [context]
   libfreenect2::Freenect2 freenect2;
-  // TODO: enable on merge
-  //libfreenect2::Freenect2Replay freenect2replay;
+  libfreenect2::Freenect2Replay freenect2replay;
   libfreenect2::Freenect2Device *dev = 0;
   libfreenect2::PacketPipeline *pipeline = 0;
 /// [context]
@@ -340,11 +339,11 @@ int main(int argc, char *argv[])

     std::vector<std::string> frame_filenames;

-    d = opendir("recordings/depth");
+    d = opendir("../recordings/depth");

     if(!d)
     {
-      std::cerr << "Could not open directory " << dir << " for replay." << std::endl;
+      std::cerr << "Could not open directory " << d << " for replay." << std::endl;
       exit(1);
     }

@@ -352,17 +351,18 @@ int main(int argc, char *argv[])
     {
       std::string name = dir->d_name;

-      if(hasSuffix(name, ".depth"))
+      if(hasSuffix(name, ".jpg"))
       {
+        //frame_filenames.push_back("../recordings/depth/" + name);
         frame_filenames.push_back(name);
+        std::cerr << "Name: " << name << std::endl;
       }
       else
       {
         std::cerr << "Skipping currently unsupported frame filename: " << name << std::endl;
       }
     }
-    // TODO: enable on merge
-/*
+
     if(pipeline)
     {
 /// [open]
@@ -373,7 +373,6 @@ int main(int argc, char *argv[])
     {
       dev = freenect2replay.openDevice(frame_filenames);
     }
-*/
   }

   if(dev == 0)
diff --git a/tools/streamer_recorder/recorder.cpp b/tools/streamer_recorder/recorder.cpp
index 89afc39..ea3a02e 100644
--- a/tools/streamer_recorder/recorder.cpp
+++ b/tools/streamer_recorder/recorder.cpp
@@ -93,12 +93,13 @@ void Recorder::record(libfreenect2::Frame* frame, const std::string& frame_type)
     // std::cout << "Run Recorder." << std::endl;
     cvMat_frame = cv::Mat(frame->height, frame->width, CV_32FC1, frame->data) / 10;
     // TODO: handle relative path + check Windows / UNIX compat.
-    oss_recordPath << "../recordings/depth/" << std::setw( 5 ) << std::setfill( '0' ) << frameID << ".depth";
+    oss_recordPath << "../recordings/depth/" << std::setw( 5 ) << std::setfill( '0' ) << "_" << timeStamps[frameID] << "_" << frameID << ".jpg";
   }
   else if (frame_type == "registered" || frame_type == "rgb")
   {
     cvMat_frame = cv::Mat(frame->height, frame->width, CV_8UC4, frame->data);
     // TODO: handle relative path + check Windows / UNIX compat.
+    //oss_recordPath << "../recordings/regist/" << std::setw( 5 ) << std::setfill( '0' ) << frameID << ".jpg";
     oss_recordPath << "../recordings/regist/" << std::setw( 5 ) << std::setfill( '0' ) << frameID << ".jpg";
     // std::cout << frame->height << ":" << frame->width << ":" << frame->bytes_per_pixel << std::endl;
   }
```
Will probably get back to it after new year's and make it a PR after more extensive testing.

xlz, 2017-12-21:
I won't look into the detail. I think it makes sense to record the raws and
replay by decoding the raws, so better decoding algorithms can be tested
and compared offline with the same raws. I'm not sure why OpenCV is used
here If only saving raws. Libfreenect2 decodes the raws without OpenCV
alright. If you mean recording decoded images, depth or rgb, I don't think
it's very useful to go through the workflow of libfreenect2. I won't mind
if that's what you're doing since it's contrib but that's what I've had in
mind about the design.