OpenKinect/libfreenect2 #950: Protonect Segfaults with cl option

willf-j, 2017-12-24:
Overview Description:
Protonect runs fine on the cpu, but seg faults with cl option. I'm on Ubuntu 16.04

I'm kind of at a loss, I've run some opencl test programs and they ran fine, namely the one linked in this: https://askubuntu.com/questions/850281/opencl-on-ubuntu-16-04-intel-sandy-bridge-cpu

output of uname -a:
 4.10.0-42-generic #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Output of clinfo:

Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 17.2.4
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 1
  Device Name                                     AMD FirePro W4100 (AMD CAPE VERDE / DRM 3.20.0 / 4.10.0-42-generic, LLVM 5.0.0)
  Device Vendor                                   AMD
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.1 Mesa 17.2.4
  Driver Version                                  17.2.4
  Device OpenCL C Version                         OpenCL C 1.1 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Max compute units                               8
  Max clock frequency                             630MHz
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              64
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 2 / 2       
    half                                                 0 / 0        (n/a)
    float                                                4 / 4       
    double                                               2 / 2        (cl_khr_fp64)
  Half-precision Floating-point support           (n/a)
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Address bits                                    64, Little-Endian
  Global memory size                              7491413604 (6.977GiB)
  Error Correction support                        No
  Max memory allocation                           1872853401 (1.744GiB)
  Unified memory for Host and Device              No
  Minimum alignment for any data type             128 bytes
  Alignment of base address                       1024 bits (128 bytes)
  Global Memory cache type                        None
  Image support                                   No
[... 101 lines cut ...]
#8  0x00007fffe3732f3f in llvm::GCNScheduleDAGMILive::schedule() () from /usr/lib/x86_64-linux-gnu/libLLVM-5.0.so.1
#9  0x00007fffe3732c7e in llvm::GCNScheduleDAGMILive::finalizeSchedule() () from /usr/lib/x86_64-linux-gnu/libLLVM-5.0.so.1
#10 0x00007fffe27d3da5 in ?? () from /usr/lib/x86_64-linux-gnu/libLLVM-5.0.so.1
#11 0x00007fffe276c991 in llvm::MachineFunctionPass::runOnFunction(llvm::Function&) () from /usr/lib/x86_64-linux-gnu/libLLVM-5.0.so.1
#12 0x00007fffe25dcac2 in llvm::FPPassManager::runOnFunction(llvm::Function&) () from /usr/lib/x86_64-linux-gnu/libLLVM-5.0.so.1
#13 0x00007fffe3159d2e in ?? () from /usr/lib/x86_64-linux-gnu/libLLVM-5.0.so.1
#14 0x00007fffe25dc70f in llvm::legacy::PassManagerImpl::run(llvm::Module&) () from /usr/lib/x86_64-linux-gnu/libLLVM-5.0.so.1
#15 0x00007fffe552eb89 in ?? () from /usr/lib/x86_64-linux-gnu/libMesaOpenCL.so.1
#16 0x00007fffe552f3b0 in ?? () from /usr/lib/x86_64-linux-gnu/libMesaOpenCL.so.1
#17 0x00007fffe552b28a in ?? () from /usr/lib/x86_64-linux-gnu/libMesaOpenCL.so.1
#18 0x00007fffe551c25f in ?? () from /usr/lib/x86_64-linux-gnu/libMesaOpenCL.so.1
#19 0x00007fffe54fa5db in ?? () from /usr/lib/x86_64-linux-gnu/libMesaOpenCL.so.1
#20 0x00007ffff7b8c1a8 in cl::Program::build (this=0x6c8178, 
    options=0x7d6820 " -D BFI_BITMASK=0x180 -D AB_MULTIPLIER=6.6666668653488159e-01f -D AB_MULTIPLIER_PER_FRQ0=1.3225810527801514e+00f -D AB_MULTIPLIER_PER_FRQ1=1.", '0' <repeats 16 times>, "e+00f -D AB_MULTIPLIER_PER_FRQ2=1.612902998"..., notifyFptr=0x0, data=0x0) at /home/rtr/libfreenect2/include/internal/CL/cl.hpp:5269
#21 0x00007ffff7b922d9 in libfreenect2::OpenCLDepthPacketProcessorImpl::buildProgram (this=0x6c8060, 
    sources="/*\n * This file is part of the OpenKinect Project. http://www.openkinect.org\n *\n * Copyright (c) 2014 individual OpenKinect contributors. See the CONTRIB file\n * for details.\n *\n * This code is licens"...) at /home/rtr/libfreenect2/src/opencl_depth_packet_processor.cpp:634
#22 0x00007ffff7b8f061 in libfreenect2::OpenCLDepthPacketProcessorImpl::initDevice (this=0x6c8060, deviceId=-1) at /home/rtr/libfreenect2/src/opencl_depth_packet_processor.cpp:454
#23 0x00007ffff7b8d519 in libfreenect2::OpenCLDepthPacketProcessorImpl::OpenCLDepthPacketProcessorImpl (this=0x6c8060, deviceId=-1)
    at /home/rtr/libfreenect2/src/opencl_depth_packet_processor.cpp:245
#24 0x00007ffff7b8aa28 in libfreenect2::OpenCLDepthPacketProcessor::OpenCLDepthPacketProcessor (this=0x6e7420, deviceId=-1)
    at /home/rtr/libfreenect2/src/opencl_depth_packet_processor.cpp:721
#25 0x00007ffff7b61ff0 in libfreenect2::OpenCLPacketPipeline::OpenCLPacketPipeline (this=0x6cfaf0, deviceId=-1) at /home/rtr/libfreenect2/src/packet_pipeline.cpp:151
#26 0x000000000040fa5e in main (argc=2, argv=0x7fffffffd9f8) at /home/rtr/libfreenect2/examples/Protonect.cpp:198

Any help would be greatly appreciated

xlz, 2017-12-25:
Probably a bug in Mesa. I found a similar one with no actual resolution https://bugs.freedesktop.org/show_bug.cgi?id=94503.

You have Mesa 17.2.4. I noticed Ubuntu 16.04 only has 17.0.7 from https://packages.ubuntu.com/search?keywords=mesa-opencl-icd. Maybe you can try different versions of Mesa.

willf-j, 2017-12-29:
thanks for the tip. I ended up getting so frustrated dealing with the AMD gpu and all the associated driver issues that I tossed it in the trash, swore never to buy an AMD gpu again, swapped in an nvidia GPU, and it works like a charm. Can probably close, but I don't have any tips for future readers.

closed 2018-11-18 by xlz