OpenKinect/libfreenect2 #951: issues about streamer_recorder

suijingfeng, 2017-12-25:
my computer system is ubuntu16.04 and have opencv3 installed by 
`sudo apt-get install ros-kinetic-opencv3`

ros-kinetic-opencv3 is already the newest version (3.3.1-0xenial-20171106-101742-0800).

so i am not install opencv3 as the README.md instructed.

1. when compiling the streamer_recorder, i meet the following problem:

make[2]: *** No rule to make target 'opencv_calib3d-NOTFOUND', needed by 'bin/Kinect2Grabber'.  Stop.
CMakeFiles/Makefile2:484: recipe for target 'tools/cvPclGrabber/CMakeFiles/Kinect2Grabber.dir/all' failed
make[1]: *** [tools/cvPclGrabber/CMakeFiles/Kinect2Grabber.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

but when I replace the line 

` SET(CMAKE_BUILD_TYPE RelWithDebInfo CACHE STRING "Choose: RelWithDebInfo Release Debug MinSizeRel None") ` with

`SET(CMAKE_BUILD_TYPE  Release)`
problem solved.

2.  OpenCV Error: Unspecified error (could not find a writer for the specified extension) in imwrite_, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgcodecs/src/loadsave.cpp, line 677
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgcodecs/src/loadsave.cpp:677: error: (-2) could not find a writer for the specified extension in function imwrite_

this problem is caused  by  `if (hasSuffix(frame, ".depth"))`
As https://github.com/OpenKinect/libfreenect2/issues/948 indicates:
OpenCV does not like .depth extension and is all fussy about what kind of image types it can save.
so the .depth suffix should be replaced or usr other image saving method.

smokhov, 2017-12-26:
@suijingfeng -- thanks for your report, though it is a bit premature... streamer_recorder is a work-in-progress as a series of PRs and #948 is our current working TODO list. I plan to get back to it after the holidays. My plan is to make a non-OpenCV build for raw recordings for it to replay. The patch in #948 allows you to save your recorded session. Other than that this issue can be closed; stay tuned in 948.

smokhov, 2018-01-05:
@suijingfeng -- it just occurred to me -- are we talking about the same thing here? In your log you mention `Kinect2Grabber`... from `cvPclGrabber`... not something is provided by `libfreenect2` despite the subject of the issue mentioning our tool. What's the relationship here with `ros-kinetic-opencv3`?

suijingfeng, 2018-01-05:
@smokhov -- sorry, the logs I provided is wrong. I have correct it, but the issue do exist. 
Kinect2Grabber and cvPclGrabber have nothing to do with you tool, but do the similar work if depended on opencv. This issues can be closed, but i will find other issues in your code!

closed 2018-01-05 by suijingfeng

parkboyoung, 2018-08-28:
hi suijingfeng! 
OpenCV Error: Unspecified error (could not find a writer for the specified extension) in imwrite_, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgcodecs/src/loadsave.cpp, line 677
terminate called after throwing an instance of 'cv::Exception'
what(): /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgcodecs/src/loadsave.cpp:677: error: (-2) could not find a writer for the specified extension in function imwrite_

I have also this problem. How did you solve it?

DavidTxx, 2018-08-29:
Got the same issue here while trying to use recorder, runnning MacOs 10.13.4:
 
./bin/ProtonectSR -recorder
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/ProtonectSR [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-recorder] [-streamer] [-replay]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 ProtonectSR
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:13 with serial 017039145047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 017039145047
device firmware: 2.3.3913.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 992
Initialize Recorder.
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 5 packets were lost
libc++abi.dylib: terminating with uncaught exception of type cv::Exception: OpenCV(3.4.2) /tmp/opencv-20180704-99354-1kt9ska/opencv-3.4.2/modules/imgcodecs/src/loadsave.cpp:684: error: (-2:Unspecified error) could not find a writer for the specified extension in function 'imwrite_'

Abort trap: 6

smokhov, 2018-08-29:
On Wed, 29 Aug 2018, DavidTxx wrote:

> Date: Wed, 29 Aug 2018 04:06:43 -0700
> 
> Got the same issue here while trying to use recorder, runnning MacOs 10.13.4:
>
> ./bin/ProtonectSR -recorder
> ...

I'll get to working on it again this end of September/October timeframe. 
It's been on my plate for a while... You may have a look at the 
corresponding patch posted in the related issue in the meantime.

-- 
Serguei Mokhov, PhD, Manager, Networks & Security, AITS | /~\    The ASCII
Affiliate @ Computer Science and Software Engineering & | \ / Ribbon Campaign
Concordia Institute for Information Systems Engineering |  X    Against HTML
Concordia University, Montreal, Quebec, Canada          | / \      Email!
http://www.cs.concordia.ca/~mokhov
---
Co-founder, CCIFF, http://cciff.ca
ET Summit: http://cciff.ca/2018/festival/et/
https://www.youtube.com/channel/UCcJ-jSP6Cya8o-aWoB4l3Pw
---
Technical Lead
http://mdreams-stage.com
http://users.encs.concordia.ca/~gipsy
http://marf.sf.net | http://sf.net/projects/marf

hanikh, 2019-03-02:
Got the same issue while trying to use recorder, my system is ubuntu 16.04.
is there any solution for that?

hanikh, 2019-03-02:
./bin/ProtonectSR -recorder
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/ProtonectSR [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-recorder] [-streamer] [-replay]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 ProtonectSR
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 125655440947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 125655440947
device firmware: 4.0.3917.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
Initialize Recorder.
OpenCV Error: Unspecified error (could not find a writer for the specified extension) in imwrite_, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgcodecs/src/loadsave.cpp, line 677
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgcodecs/src/loadsave.cpp:677: error: (-2) could not find a writer for the specified extension in function imwrite_

Aborted (core dumped)