OpenKinect/libfreenect2 #952: make failed on macOS: nvcc fatal   : The version ('90000') of the host compiler ('Apple clang') is not supported

karlapps, 2017-12-29:
```
rmbp: build alur$ make
[  6%] Built target generate_resources_tool
[  9%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/src/cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o
nvcc warning : The 'compute_20', 'sm_20', and 'sm_21' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
nvcc fatal   : The version ('90000') of the host compiler ('Apple clang') is not supported
CMake Error at cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o.RelWithDebInfo.cmake:219 (message):
  Error generating
  /Users/alur/Documents/kinect/libfreenect2/build/CMakeFiles/cuda_compile_1.dir/src/./cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o

make[2]: *** [CMakeFiles/cuda_compile_1.dir/src/cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2
```

heroalur, 2017-12-29:
Same issue.

xlz, 2017-12-30:
http://docs.nvidia.com/cuda/pdf/CUDA_Toolkit_Release_Notes.pdf says Xcode 9.1 is not supported by CUDA now.

After you change clang's version, you need to rm -rf build and cmake .. from scratch. If the problem still persists, we shall look into how nvcc finds and calls clang.

heroalur, 2017-12-30:
@xlz 感谢您的回答. 稍后试试.

closed 2018-01-30 by xlz