OpenKinect/libfreenect2 #953: Depth values not working properly.

ferasboulala, 2018-01-01:
Hi there.

I have recently acquired a Kinect v2. I downloaded and installed ```libfreenect2```. ```Protonect``` works just fine, I can use a similar code to access images from the device and store them into ```Mat``` objects.

My goal is to simply do facial detection with either the ```rgb``` or ```registered```. I managed to do the following:
1. Detect a face in ```rgb``` and ```registered```;
2. Show the image with a square around the face;

Now, I would like to get the depth aka the distance of the face from the kinect. Knowing this, I decided to use the ```registered``` image since the ```rgb``` is not naturally aligned with the ```depth``` image. After making sure the calibration values were sufficient, I simply took the center of the face, ```(x,y)```, and took the average of a 20 pixels box around it ```(x+-10,y+-10)```. The problem is that the values don't make any sense. It jumps from 600 mm to over 2000. When I move the face to the right plane of the image, all values become 0. 

Here are screenshots for reference:
This is a good distance read. 
[image]

Simply moving the picture around gives me a huge jumps. The picture is not at 2 meters from the kinect.
[image]

This one is self-explanatory. 
[image]

Any ideas?

xlz, 2018-01-01:
I think there is a bug in your code.

> the average of a 20 pixels box around it

But why? Why not just take the center? You can print the depth values of those pixels and should find some of them have unexpected values.

ferasboulala, 2018-01-01:
Thank you for your response.

I have chosen to use an average of neighboring pixels in the case I end up on a dead pixel or a wrong value. I have tried to compute only the middle pixel, but the problem seems to persist. I printed 25 pixels (5x5) around the center and they all share similar (and wrong) values. It is very weird that ```imshow()``` displays a normal looking image but when I want to actually have those values, they don't make any sense. Here is a snippet of my code if I simply consider the pixel at the middle:

`listener.waitForNewFrame(frames);`
`libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];`
`libfreenect2::Frame *depth = frames[libfreenect2::Frame::Depth];`
`registration->apply(rgb, depth, &undistorted, &registered, true, &depth2rgb);`

`Mat(rgb->height, rgb->width, CV_8UC4, rgb->data).copyTo(rgbmat);`
`Mat(depth->height, depth->width, CV_32FC1, depth->data).copyTo(depthmat);`
`Mat(registered.height, registered.width, CV_8UC4, registered.data).copyTo(rgbd);`

`ostringstream distance;`
`face_cascade.detectMultiScale(rgbd, faces, 1.1, 2, 0|CASCADE_SCALE_IMAGE, Size(30, 30));`

`for (uint i = 0; i < faces.size(); i++) {`
--------`rectangle(depthmat, faces[i], Scalar(255), 2);`
--------`rectangle(rgbd, faces[i], Scalar(255,0,0), 2);`
--------`distance << depthmat.at<float>(faces[i].x + int(faces[i].width/2), faces[i].y + int(faces[i].height/2));`
--------`cout << distance.str() << endl;`
--------`putText(rgbd, distance.str(), Point(faces[i].x, faces[i].y), cv::FONT_HERSHEY_SIMPLEX, 1, Scalar(255,255,255), 1, LINE_AA);`
`}`

`imshow("depth", depthmat / 4500.0f);`
`imshow("registered", rgbd);`

`face_cascade()` fills the `faces` vector with `Rect` structures. It contains `(x,y)` values and the corresponding height and width.

ferasboulala, 2018-01-02:
I figured it out. As always, stupid mistake... at<float> should access the coordinates in this order instead: (y,x). OpenCV confused me a little bit when you instance a Scalar() or a Mat() object. Thank you for your help. Library works great so far. Very much appreciated.

closed 2018-01-02 by ferasboulala