OpenKinect/libfreenect2 #955: LibUSB : Git auto-merging not longer working in install_libusb_vs%YEAR%.cmd

AlexMorgand, 2018-01-04:
Due to recents commits in the https://github.com/libusb/libusb git repository, the auto merge of the branch **xlz** and **winiso** is no longer working (https://github.com/libusb/libusb/commits/master) caused by the line in the file **install_libusb_vs%YEAR%.cmd** :

> git merge xlz/winiso

This results in : 

> Auto-merging libusb/os/windows_winusb.h
> CONFLICT (content): Merge conflict in libusb/os/windows_winusb.h
> Auto-merging libusb/os/windows_winusb.c
> CONFLICT (content): Merge conflict in libusb/os/windows_winusb.c
> Automatic merge failed; fix conflicts and then commit the result.
> ....
>    22 Warning(s)
>    102 Errors(s)

You can do the merge by hand in the files **\libusb\os\windows_winusb.c** and **\libusb\os\windows_winusb.h**  if you need to fix it asap but I think a quick update on the branch xlz will fix the issue.

xlz, 2018-01-05:
Yeah, it's a problem. They wouldn't merge the patches for years https://github.com/libusb/libusb/pull/284 https://github.com/libusb/libusb/pull/246. I'll do this later.

xlz, 2018-01-05:
I pushed fixed patches to xlz/winiso. Does it work now?

AlexMorgand, 2018-01-05:
Got some missing symbols : 

> "...\msvc\libusb_dll_2013.vcxproj" (Rebuild cible) (1) ->
> (Link cible) ->
>   windows_winusb.obj : error LNK2001: unresolved external symbol unsupported_submit_iso_transfer 
>   windows_winusb.obj : error LNK2001: unresolved external symbol unsupported_copy_transfer_data 
>   libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.dll : fatal error LNK1120: 2 unresolved externals

xlz, 2018-01-06:
These symbols were removed some time ago. I rebased the patches again, please test build.

AlexMorgand, 2018-01-06:
:+1:  Works for me :) Thank you

closed 2018-01-06 by xlz