OpenKinect/libfreenect2 #956: bytes per channel RGB image

fred-dev, 2018-01-08:
I am trying to use the` registration->apply()` function, I am passing the correct frames as far as I can tell. My rgb image comes from
`
        libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];`

my 	rgb->bytes_per_pixel is checked as 3, however the registration->apply() function expects this to be 4, so the apply fails. I would expect this to work as is, but maybe I am wrong. How can I end up with an rgb frame of 4 bytes per pixel?

xlz, 2018-01-08:
bytes_per_pixel should not be 3. The library doesn't set it to 3 anywhere AFAICT. What does your code look like?

fred-dev, 2018-01-08:
```
int ofProtonect::openKinect(string serial){
          
        pipeline = new libfreenect2::OpenCLPacketPipeline(1);

      if(pipeline)
      {
        dev = freenect2.openDevice(serial, pipeline);
      }

      if(dev == 0)
      {
        ofLogError("ofProtonect::openKinect")  << "failure opening device with serial " << serial;
        return -1;
      }
      listener = new libfreenect2::SyncMultiFrameListener(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
      dev->setColorFrameListener(listener);
      dev->setIrAndDepthFrameListener(listener);
      dev->start();
      registration = new libfreenect2::Registration(dev->getIrCameraParams(), dev->getColorCameraParams());
	
	bOpened = true;
    return 0;
}
```
Then I call my own update loop:

```
void ofProtonect::updateKinect(ofPixels & rgbPixels, ofFloatPixels & depthPixels){
  
    if(bOpened){
        listener->waitForNewFrame(frames);
        libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];
        libfreenect2::Frame *ir = frames[libfreenect2::Frame::Ir];
        libfreenect2::Frame *depth = frames[libfreenect2::Frame::Depth];

		libfreenect2::Frame undistorted(512, 424, 4), registered(512, 424, 4);
		bigDepth = new libfreenect2::Frame(1920, 1080 + 2, 3);
		
        rgbPixels.setFromPixels(rgb->data, rgb->width, rgb->height, 3);
        depthPixels.setFromPixels((float *)depth->data, ir->width, ir->height, 1);

		cout << std::to_string(rgb->bytes_per_pixel) << endl;
		
		registration->apply(rgb, depth, &undistorted, &registered, false);

        listener->release(frames);
    }
 }
```
The line:
	`	cout << std::to_string(rgb->bytes_per_pixel) << endl;
`Is printing 3.

xlz, 2018-01-08:
```
bigDepth = new libfreenect2::Frame(1920, 1080 + 2, 3);
                                                   ^
```

closed 2018-01-30 by xlz