OpenKinect/libfreenect2 #958: LED does not always turn off?

svenevs, 2018-01-10:
`git log -1 --oneline`: [`05beb27`](https://github.com/OpenKinect/libfreenect2/commit/05beb277bb4018ff25a24c90beec8949b09a2ac9) windows: Build upstream libusb only`

xlz, 2018-01-11:
I don't know exactly. My best guess is the white LED indicates if the Kinect's ARM board is running. As long as it is not emitting IR I think this is okay for the sensors. I mean, how do you communicate with the Kinect if it is powered off?

We don't know anything at the protocol level about controlling the LEDs per se. We do use a ShutdownCommand only on Mac https://github.com/OpenKinect/libfreenect2/blob/ec47090673b7d940741dace33e4d4fbe0b32e265/src/libfreenect2.cpp#L972 for some issue. We don't know what exactly this command does though. I think the reason it is not used on other platforms is because "this command actually reboots the device and makes it disappear for 3 seconds" (so it's probably better named RebootCommand) and we want that Protonect can restart instantly.

svenevs, 2018-01-11:
LOL yeah rebooting by default is best left only to when it's needed!

> I mean, how do you communicate with the Kinect if it is powered off?

I think it's always on if it's plugged in, regardless of the white LED.  If you leave it plugged in but don't do anything with it, no LEDs on, etc, it still generates heat.  I wouldn't be surprised if the terms and conditions allow the video feed to be activated at any time (paranoid)...

Anyway, I agree -- this shouldn't be harfmul to the device and it kind of looks cool.  Thanks for the response :)

closed 2018-01-11 by svenevs