OpenKinect/libfreenect2 #962: Kinect One locks USB devices

digitalmeltingculture, 2018-01-21:
When I switch the PC on with the Kinect already plugged on, the other USB devices (mouse, keyboard, etc...) do not work.

I cannot manually plug the Kinect after the PC boot. I need to have the Kinect already plugged on when the PC boots. Does anyone have the same need?
How could I solve this?
I already tried to use an external powered USB switch but it did not solve the issue (i used an ADJ USB 3.0 powered hub).

xlz, 2018-01-23:
Your OS? Any logs?

digitalmeltingculture, 2018-01-23:
The OS is Ubuntu Studio. I had the same problem both on 16.04 and 17.04.
I have not any log about that. How could I log the USB ports behaviour?

digitalmeltingculture, 2018-01-27:
I also tried to set the USB autosuspend to -1 as suggested in the specifics, but it does not improve the status.
@xlz how could I report the logs you asked?

xlz, 2018-01-30:
Duplicate of #940.

closed 2018-01-30 by xlz