OpenKinect/libfreenect2 #963: Problems with usb 3.0

KoSv, 2018-01-24:
Hello everybody. I'm trying to connect my kinect2 to my Ubuntu 16.04
I've connected my kinect v2 with an external power supply (http://www.tatsch.it/kinect-2-usb-3-modification/).
Installed libfreenect2. (I have also libfreenect1. should I uninstall it?)
but the device doesnt' show up.
this is what ive got:
lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 056a:00b9 Wacom Co., Ltd Intuos4 6x9
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 065: ID 0781:5581 SanDisk Corp. Ultra
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

but here it is but there is a problem I cannot solve..
dmesg | egrep -i '(dma|usb)':

usb 9-2: new SuperSpeed USB device number 97 using xhci_hcd
[ 1248.797493] usb 9-2: New USB device found, idVendor=045e, idProduct=02c4
[ 1248.797506] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 1248.797514] usb 9-2: Product: Xbox NUI Sensor
[ 1248.797520] usb 9-2: Manufacturer: Microsoft
[ 1248.797525] usb 9-2: SerialNumber: 382431733947
[ 1249.656257] usb usb9-port2: cannot disable (err = -32)
[ 1249.656269] usb 9-2: USB disconnect, device number 97

any idea?
greetz, KoSv

xlz, 2018-01-24:
This is purely a USB driver problem. Try a newer kernel. Otherwise we can't help you.

KoSv, 2018-01-24:
I got it! It was the power supply! an 1100mA. and perhaps somehow faulty..
Now it works!

closed 2018-01-30 by xlz