OpenKinect/libfreenect2 #965: [question] millimeters, bilateral, and intrinsics/extrinsics

svenevs, 2018-02-09:
`git log -1 --oneline`: 05beb27 windows: Build upstream libusb only

I was digging around the internals and had three questions that I believe all have clear answers.

1. Are the units meters or millimeters for the depth feed?  The [docs say millimeters](https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Frame.html#a1d1cfd8ffb84e947f82999c682b666a7af5c0a1b2ee6f0c691d602752b185463c), but in my own code I have to divide by `1000.0f` to get reasonable results.  I also noticed this line

    https://github.com/OpenKinect/libfreenect2/blob/05beb277bb4018ff25a24c90beec8949b09a2ac9/src/libfreenect2.cpp#L710

    That would make sense for meters, not millimeters right?  I've traced my own code many times, and cannot find anything blatant that would require me to divide by `1000.0f` unless the units were actually meters.  Just looking for confirmation of units.

2. Regardless of the pipeline choice, bilateral filtering is **on** by default?

    https://github.com/OpenKinect/libfreenect2/blob/05beb277bb4018ff25a24c90beec8949b09a2ac9/src/libfreenect2.cpp#L711

    I just found out about this, and it means I'm filtering twice!

3. I've read through #41 and related #144 and the [registration docs](https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html) a few times, but I'm a little confused about the (awesome) hackery that went on.  The synopsis is that mapping depth -> color is **more reliable** than color -> depth, correct?  I understand that calibrating the cameras is ideal, just trying to make sure I understand the state of affairs.  (To be extra clear, I'm not complaining, the fact that this is even working given what had to be done to figure it out is impressive!)

Thanks for any insight!

xlz, 2018-02-20:
> 1. Are the units meters or millimeters for the depth feed?

The configs are in meters. https://openkinect.github.io/libfreenect2/structlibfreenect2_1_1Freenect2Device_1_1Config.html#a6f421967e54ce6a90b0a63323557da9d

This is inconsistency indeed, but it is left alone for the sake of not changing working things. 

Millimeter has a better precision than meter for internal calculation. It's probably better to leave it be than to divide by 1000 again for output.

> 2. Regardless of the pipeline choice, bilateral filtering is on by default?

It is. Is it a problem? I'm not aware of any secondary bilateral filtering going on here.

> 3. mapping depth -> color is more reliable than color -> depth, correct?

The depth to color (given a 3D point represented by a 2D coordinate in the depth image and the depth value, find the 2D coordinate of the 3D point in the color image) mapping formula (`kinect2_depth_to_color` in #41) is reverse engineered and should be reliable as in it is official (it is certainly no more reliable than actual calibration.) The reverse direction was too difficult to reverse engineer and the method we use is some kind of Reverse Pixel Mapping, see http://www.imagemagick.org/Usage/distorts/#mapping.

svenevs, 2018-04-03:
Hey @xlz,

Thanks for the response, sorry for not responding!

> It is. Is it a problem? I'm not aware of any secondary bilateral filtering going on here.

It's a problem for my framework that sits on top of `libfreenect2`, not a problem with `libfreenect2`.  I was using the bilateral filter in my own code as well, meaning that for Kinect v2 devices it was getting filtered twice.  I'd be happy to submit a brief docs PR that brings this detail into focus if you like (and indicate where the documentation should go).  I found this by accident looking through the code, I think that it would be nice to inform users that bilateral filtering happens by default so they know not to filter themselves.

Thank you again for answering my questions, sorry this "issue" remained open needlessly for so long!

closed 2018-04-03 by svenevs