OpenKinect/libfreenect2 #967: libUSB and USBDK

gdardenn, 2018-02-16:
Hi,

I have a problem concerning the install of libUSB with libfreenect2.
I work on Windows 10 (x64) with Visual 2015 (community).
I launched "install_libusb_vs2015.cmd" in order to install libusb.

I have an error concerning the build of libusb_usbdk_dll_2015.

U:\01-Development\ThirdParties\libfreenect2\depends\libusb_src>"C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe" msvc\libusb_usbdk_dll_2015.vcxproj /p:Platform=x64 /p:Configuration=Release /target:Rebuild   || exit /b
Microsoft (R) Build Engine, version 14.0.25420.1
Copyright (C) Microsoft Corporation. Tous droits réservés.

MSBUILD : error MSB1009: The file does not exist
Commutateur : msvc\libusb_usbdk_dll_2015.vcxproj

Why this target is not copied ? Do I need to modify some options concerning libusb ?
If yes, how ?

Thanks,
Guillaume DARDENNE

gdardenn, 2018-02-16:
Ok, when I make a git reset to a previous version of libusb (for instance with the commit hash "d0779e9", it works.
I added the following line to test into install_libusb_vs2015.cmd :
git reset --hard d0779e9

xlz, 2018-02-20:
I think this is fixed by #969.

closed 2018-02-20 by xlz