OpenKinect/libfreenect2 #968: Cannot Install OpenNi2 Mac OS

Tenkir, 2018-02-19:
Overview Description:
```Error: No available formula with the name "openni2"
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/core is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
==> Searching local taps...
This similarly named formula was found:
totakke/openni2/openni2-freenectdriver
To install it, run:
  brew install totakke/openni2/openni2-freenectdriver
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.```

Seems openni2 is missing from Homebrew since the deprecation of the `homebrew/science` tap.

xlz, 2018-02-20:
Maybe the instructions need an update? What is the current way to install openni2 with homebrew?

Tenkir, 2018-02-22:
I don't believe there is one, at least based on what I've found.

sushilei, 2018-03-01:
$ brew tap brewsci/science
and then
$ brew install openni2

commit e38cf32a (2018-03-01): docs: Update OpenNI2 tap for Mac OS

Fix #968.
closed 2018-03-01 by xlz