OpenKinect/libfreenect2 #971: Protonect works for single Kinect v2 but shows following error for 2 Kinect v2: Not enough bandwidth for new device state

ravijo, 2018-02-22:
I am trying to use multiple Kinect in Ubuntu 14.04 LTS workstation. Unfortunately, `Protonect` works for single Kinect v2 but shows following error for 2 Kinect v2: 
`Not enough bandwidth for new device state`

[Troubleshooting](https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#check-dmesg-after-running-protonect) page shows following-

- **Not enough bandwidth for new device state**: Your hardware does not have required bandwidth for the Kinect, even if it supports USB 3 

I am wondering since in my case, `Protonect` is working fine for a single Kinect v2 sensor. but shows errors when two Kinect v2 are plugged in.

What is the culprit here? Any suggestions, please?

-------------------------------
Below are the required logs-
```
ravi@baxter:~/tools/libfreenect2/build/bin$ export LIBUSB_DEBUG=3
ravi@baxter:~/tools/libfreenect2/build/bin$ ./Protonect 510428142542
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:7 with serial 510428142542
[Info] [Freenect2Impl] found valid Kinect v2 @2:6 with serial 510760242542
[Info] [Freenect2Impl] found 2 devices
[Info] [Freenect2DeviceImpl] opening...
libusb: error [op_get_configuration] device unconfigured
libusb: error [op_set_configuration] failed, error -1 errno 28
[Error] [protocol::UsbControl] failed to set configuration! LIBUSB_ERROR_OTHER Other error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:7
failure opening device!
```
The `uname -a` returns following message-
`Linux baxter 4.2.0-42-generic #49~14.04.1-Ubuntu SMP Wed Jun 29 20:22:11 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

The `lsusb -t` returns following message-
```
ravi@baxter:~/tools/libfreenect2/build/bin$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
    |__ Port 1: Dev 4, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 5, If 0, Class=Hub, Driver=hub/1p, 5000M
            |__ Port 1: Dev 6, If 0, Class=Vendor Specific Class, Driver=usbfs, 5000M
            |__ Port 1: Dev 6, If 1, Class=Vendor Specific Class, Driver=usbfs, 5000M
            |__ Port 1: Dev 6, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 6, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 5: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 4, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 9: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 10: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```
`lspci` returns following information-
```
ravi@baxter:~/tools/libfreenect2/build/bin$ lspci
00:00.0 Host bridge: Intel Corporation Device 591f (rev 05)
00:01.0 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x16) (rev 05)
00:14.0 USB controller: Intel Corporation Device a2af
00:14.2 Signal processing controller: Intel Corporation Device a2b1
00:16.0 Communication controller: Intel Corporation Device a2ba
00:17.0 SATA controller: Intel Corporation Device a282
00:1b.0 PCI bridge: Intel Corporation Device a2e9 (rev f0)
00:1c.0 PCI bridge: Intel Corporation Device a292 (rev f0)
00:1c.6 PCI bridge: Intel Corporation Device a296 (rev f0)
00:1c.7 PCI bridge: Intel Corporation Device a297 (rev f0)
00:1d.0 PCI bridge: Intel Corporation Device a298 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device a2c4
00:1f.2 Memory controller: Intel Corporation Device a2a1
00:1f.3 Audio device: Intel Corporation Device a2f0
00:1f.4 SMBus: Intel Corporation Device a2a3
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (2) I219-V
01:00.0 VGA compatible controller: NVIDIA Corporation Device 1c82 (rev a1)
01:00.1 Audio device: NVIDIA Corporation Device 0fb9 (rev a1)
04:00.0 PCI bridge: ASMedia Technology Inc. ASM1083/1085 PCIe to PCI Bridge (rev 03)
06:00.0 Ethernet controller: Intel Corporation 82574L Gigabit Network Connection
```
The last few lines of `dmesg` are interesting. Please see below-
```
[65312.832892] usb 2-5.1: Not enough bandwidth for altsetting 1
[65312.833324] usb 2-5.1: 3:1: usb_set_interface failed (-28)
[65501.353136] usb 2-1.1.1: reset SuperSpeed USB device number 6 using xhci_hcd
[65501.427191] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 8 ep 2 with no TDs queued?
[65501.428916] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 8 ep 2 with no TDs queued?
[65547.923963] usb 2-5.1: reset SuperSpeed USB device number 3 using xhci_hcd
[65547.940437] usb 2-5.1: Not enough bandwidth for new device state.
[65547.940442] usb 2-5.1: Busted HC?  Not enough HCD resources for old configuration.
[65547.942567] usb 2-5.1: USB disconnect, device number 3
[65547.942720] usb 2-5.1: Not enough bandwidth for new device state.
[65547.942722] usb 2-5.1: Not enough bandwidth for altsetting 1
[65547.943600] systemd-udevd[3287]: Failed to apply ACL on /dev/snd/controlC2: No such file or directory
[65547.943604] systemd-udevd[3287]: Failed to apply ACL on /dev/snd/controlC2: No such file or directory
[65547.943610] systemd-udevd[3288]: Failed to apply ACL on /dev/snd/pcmC2D0c: No such file or directory
[65547.943615] systemd-udevd[3288]: Failed to apply ACL on /dev/snd/pcmC2D0c: No such file or directory
[65548.601089] usb 2-5.1: new SuperSpeed USB device number 7 using xhci_hcd
[65548.617593] usb 2-5.1: New USB device found, idVendor=045e, idProduct=02d8
[65548.617596] usb 2-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[65548.617597] usb 2-5.1: Product: Xbox NUI Sensor
[65548.617599] usb 2-5.1: Manufacturer: Microsoft
[65548.617600] usb 2-5.1: SerialNumber: 510428142542
[65548.618046] usb 2-5.1: Not enough bandwidth for new device state.
[65548.618053] usb 2-5.1: can't set config #1, error -28
[65548.945935] usb 2-5.1: Not enough bandwidth for new device state.
[65597.480522] usb 2-5.1: reset SuperSpeed USB device number 7 using xhci_hcd
[65597.497062] usb 2-5.1: Not enough bandwidth for new device state.
[65639.310076] usb 2-5.1: reset SuperSpeed USB device number 7 using xhci_hcd
[65639.326738] usb 2-5.1: Not enough bandwidth for new device state.
[66030.060202] usb 2-5.1: reset SuperSpeed USB device number 7 using xhci_hcd
[66030.076867] usb 2-5.1: Not enough bandwidth for new device state.
[66045.626549] usb 2-5.1: reset SuperSpeed USB device number 7 using xhci_hcd
[66045.643200] usb 2-5.1: Not enough bandwidth for new device state.
```

Please note that I have already increased `usbfs_memory`. See below-
```
tom@baxter:~$ cat /etc/default/grub | grep GRUB_CMDLINE_LINUX_DEFAULT
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=128"
```

lesterlo, 2018-02-22:
Did you use separate USB interface cards for your Kinects?
You need 2 USB interface cards for 2 Kinects.

xlz, 2018-02-23:
Two Kinects require twice the bandwidth, duh. Yeah, add more PCIe USB extension cards.

ravijo, 2018-02-24:
> you plug all of the kinects in the same USB root

Thank you very much @lesterlo. Sorry, I couldn't understand it. There are many USB3.0 ports on the motherboard. I tried plugging both Kinect in those ports but didn't work.

> add more PCIe USB extension cards

Thanks, @xlz. I had a spare USB 3.0 PCI express card, which I plugged in later. It worked. However, I am wondering, what if I want to use 3 Kinect v2 Cameras?

Do you mean that **Number of USB3.0 PCI express cards = The number of Kinect v2** ?

ravijo, 2018-02-24:
@SirDifferential: Thank you very much. I should order that card.

lesterlo, 2018-02-25:
> Do you mean that Number of USB3.0 PCI express cards = The number of Kinect v2 

@ravijo  : In Short, yes.

In your case you need to use 2 kinects, so you need to buy one more USB3.0 PCI-E card. You need to plug the first kinect to your motherboard USB3.0 port and the second kinect to your PCI-E card SUB3.0 port.

smokhov, 2018-02-26:
On Sun, 25 Feb 2018, Lester Lo wrote:

> @ravijo  : In Short, yes.
>
> In your case you need to use 2 kinects, so you need to buy one more
> USB3.0 PCI-E card. You need to plug the first kinect to your motherboard
> USB3.0 port and the second [...]

FWIW, I had a success running 3 kinects with front and back USB3 ports on 
the same motherboard and a single additional USB3.0 PCI-E card in this 
setup:

https://doi.org/10.1145/2945078.2945131

ravijo, 2018-02-26:
Thank you very much.

closed 2018-02-26 by ravijo

thefakhreddin, 2021-05-18:
I have this exact issue with only one Kinect v2 connected. I'm connecting Kinect v2 through USB 3 to my laptop (Asus n580vd) on ubuntu 20. Any suggestions?

ravijo, 2021-05-18:
@thefakhreddin 

Check `dmesg` please. In general `Not enough bandwidth for new device state` should not come in the case of a single Kinect v2 device. It may be possible that there is another USB3 device connected you the motherboard. Please unplug it, if so.

Furthermore, please check `usbfs_memory` and increase it a bit.