OpenKinect/libfreenect2 #972: OpenNI2 driver resolution mismatch

eruffaldi, 2018-02-28:
Overview Description:

DepthStream supports 640 and 512
ColorStream supports 1920 and 512

We should have 1920 in DepthStream and, if needed, 640 in ColorStream. 

Working on that on my fork

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
f46fa65 (HEAD -> master, origin/master, origin/HEAD) usb: Select UsbDk with libusb_set_option
2. `uname -a`
Darwin I.local 16.7.0 Darwin Kernel Version 16.7.0: Thu Jan 11 22:59:40 PST 2018; root:xnu-3789.73.8~1/RELEASE_X86_64 x86_64
3. `lsusb -t`
not needed
4. `lspci -nn`
not needed

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2018-03-01:
Yes they are indeed not supported, unfortunately.

closed 2018-03-01 by xlz

eruffaldi, 2018-03-02:
I just meant to add a way to highlight incompatible combinations. My students had bad CUDA or CPU memory errors due to this very reason. Oh yes, people are not reading the manuals

xlz, 2018-03-02:
Supported resolutions:

Registration is controlled by ONI_DEVICE_PROPERTY_IMAGE_REGISTRATION.

IR stream is always unregistered 512x424 uint16.

* Registered, depth 640x480 uint16, color 512x424 RGB888
* Registered, depth 512x424 uint16, color 512x424 RGB888
* Unregistered, depth 640x480 uint16, color 1920x1080 RGB888

I don't have a good place to put this doc.