OpenKinect/libfreenect2 #974: libusb-1.0.lib was not built

grrrwaaa, 2018-03-14:
Overview Description:

Build libusb as described in the readme at https://github.com/OpenKinect/libfreenect2

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`: e38cf32 (HEAD -> master, origin/master, origin/HEAD) docs: Update OpenNI2 tap for Mac OS
2. `uname -a`: Win 10 Pro 64. VS Community 2015 and VS Community 2017 are both installed. Cmake 3.11 installed. 

Steps to Reproduce:

1. Cloned https://github.com/OpenKinect/libfreenect2.git
2. Installed UsbDk driver via installer at link provided
3. Ran install_libusb_vs2015.cmd

Actual Results:

libusb-1.0.dll and libusb-1.0.pdb are created and copied to /depends/libusb, but libusb-1,0.lib is not generated or copied. This seems to break the Cmake generation in the next steps of the install.

Expected Results:

the dll, pdb, and lib are generated and copied.

Reproducibility:

Tried a fresh clone, happened again. 

Additional Information:

Command line output:

```
install_libusb_vs2015.cmd

E:\GitHub\libfreenect2\depends>rem This can only be run in a Git Shell or similar environments

E:\GitHub\libfreenect2\depends>rem with access to git.exe and msbuild.exe.

E:\GitHub\libfreenect2\depends>rmdir /s /q libusb_src libusb
libusb_src\x64\Release - The directory is not empty.

E:\GitHub\libfreenect2\depends>git clone https://github.com/libusb/libusb.git libusb_src   || exit /b
fatal: destination path 'libusb_src' already exists and is not an empty directory.

E:\GitHub\libfreenect2\depends>install_libusb_vs2015.cmd

E:\GitHub\libfreenect2\depends>rem This can only be run in a Git Shell or similar environments

E:\GitHub\libfreenect2\depends>rem with access to git.exe and msbuild.exe.

E:\GitHub\libfreenect2\depends>rmdir /s /q libusb_src libusb
The system cannot find the file specified.
The system cannot find the file specified.

E:\GitHub\libfreenect2\depends>git clone https://github.com/libusb/libusb.git libusb_src   || exit /b
Cloning into 'libusb_src'...
remote: Counting objects: 13006, done.
remote: Compressing objects: 100% (43/43), done.
remote: Total 13006 (delta 81), reused 91 (delta 65), pack-reused 12888
Receiving objects: 100% (13006/13006), 3.49 MiB | 1.35 MiB/s, done.
Resolving deltas: 100% (9413/9413), done.

E:\GitHub\libfreenect2\depends>cd libusb_src

E:\GitHub\libfreenect2\depends\libusb_src>set CONFIG=Release

E:\GitHub\libfreenect2\depends\libusb_src>set YEAR=2015

E:\GitHub\libfreenect2\depends\libusb_src>set MSBUILD="C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe"

E:\GitHub\libfreenect2\depends\libusb_src>"C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe" msvc\libusb_dll_2015.vcxproj /p:Platform=x64 /p:Configuration=Release /target:Rebuild   || exit /b
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 2018-03-14 3:57:23 PM.
Project "E:\GitHub\libfreenect2\depends\libusb_src\msvc\libusb_dll_2015.vcxproj" on node 1 (Rebuild target(s)).
CoreClean:
  Creating directory "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\".
PrepareForBuild:
  Creating directory "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0 (dl
  l).tlog\".
InitializeBuildStatus:
  Creating "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0 (dll).tlog\un
  successfulbuild" because "AlwaysCreate" was specified.
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\CL.exe /c /I. /I..\libusb /Zi /nologo /W4 /WX- /
  O2 /GL /D NDEBUG /D WINVER=0x0501 /D _WIN32_WINNT=0x0501 /D _LIB /D _CRT_SECURE_NO_WARNINGS /D _WINDLL /D _UNICODE /D
   UNICODE /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"E:\GitHub\libfreenect2\depends\libusb
  _src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\\" /Fd"E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dl
  l\libusb-1.0 (dll)\vc140.pdb" /Gd /TC /errorReport:queue ..\libusb\core.c ..\libusb\descriptor.c ..\libusb\hotplug.c
  ..\libusb\io.c ..\libusb\os\poll_windows.c ..\libusb\strerror.c ..\libusb\sync.c ..\libusb\os\threads_windows.c ..\li
  busb\os\windows_nt_common.c ..\libusb\os\windows_usbdk.c ..\libusb\os\windows_winusb.c
  core.c
  descriptor.c
  hotplug.c
  io.c
  poll_windows.c
  strerror.c
  sync.c
  threads_windows.c
  windows_nt_common.c
  windows_usbdk.c
  windows_winusb.c
ResourceCompile:
  C:\Program Files (x86)\Windows Kits\8.1\bin\x86\rc.exe /D _UNICODE /D UNICODE /l"0x0409" /nologo /fo"E:\GitHub\libfre
  enect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0.res" ..\libusb\libusb-1.0.rc
Link:
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\link.exe /ERRORREPORT:QUEUE /OUT:"E:\GitHub\libf
  reenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.dll" /NOLOGO kernel32.lib user32.lib gdi32.lib winspoo
  l.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /MANIFEST /MANIFE
  STUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /DEBUG /PDB:"E:\GitHub\libfreenect2\depends\libusb_src\msv
  c\..\x64\Release\dll\libusb-1.0.pdb" /LTCG:incremental /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"E:\GitHub\libfreenect
  2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.lib" /MACHINE:X64 /DLL "E:\GitHub\libfreenect2\depends\libusb
  _src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0.res"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\core.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\descriptor.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\hotplug.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\io.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\poll_windows.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\strerror.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\sync.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\threads_windows.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\windows_nt_common.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\windows_usbdk.obj"
  "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\windows_winusb.obj"
  Generating code
  All 311 functions were compiled because no usable IPDB/IOBJ from previous compilation was found.
  Finished generating code
  libusb_dll_2015.vcxproj -> E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.dll
  libusb_dll_2015.vcxproj -> E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.pdb (Full PDB
  )
FinalizeBuildStatus:
  Deleting file "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0 (dll).tl
  og\unsuccessfulbuild".
  Touching "E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0 (dll).tlog\li
  busb-1.0 (dll).lastbuildstate".
Done Building Project "E:\GitHub\libfreenect2\depends\libusb_src\msvc\libusb_dll_2015.vcxproj" (Rebuild target(s)).

Build succeeded.
    0 Warning(s)
    0 Error(s)

Time Elapsed 00:00:04.38

E:\GitHub\libfreenect2\depends\libusb_src>mkdir ..\libusb\include\libusb-1.0

E:\GitHub\libfreenect2\depends\libusb_src>copy libusb\libusb.h ..\libusb\include\libusb-1.0
        1 file(s) copied.

E:\GitHub\libfreenect2\depends\libusb_src>mkdir ..\libusb\MS64\dll

E:\GitHub\libfreenect2\depends\libusb_src>copy x64\Release\dll\*.lib ..\libusb\MS64\dll
x64\Release\dll\*.lib
The system cannot find the file specified.
        0 file(s) copied.

E:\GitHub\libfreenect2\depends\libusb_src>copy x64\Release\dll\*.dll ..\libusb\MS64\dll
x64\Release\dll\libusb-1.0.dll
        1 file(s) copied.

E:\GitHub\libfreenect2\depends\libusb_src>copy x64\Release\dll\*.pdb ..\libusb\MS64\dll
x64\Release\dll\libusb-1.0.pdb
        1 file(s) copied.

```

grrrwaaa, 2018-03-14:
I notice that the /msvc folder also contains libusb_static_2015.vcproj files etc. Opening this and building the Release x64 target generated the lib file needed, though into a different location. Seems like this was a change in the libusb repository. 

I recommend libfreenect2 updates the "install_libusb" scripts to either

a) Checkout a specific revision of the libusb github, rather than grabbing from head (which could break dependencies in this way), or

b) add the needed lines for the static lib:

```
rem This can only be run in a Git Shell or similar environments
rem with access to git.exe and msbuild.exe.

rmdir /s /q libusb_src libusb

git clone https://github.com/libusb/libusb.git libusb_src || exit /b
cd libusb_src

set CONFIG=Release
set YEAR=2015
set MSBUILD="C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe"

%MSBUILD% msvc\libusb_dll_%YEAR%.vcxproj /p:Platform=x64 /p:Configuration=%CONFIG% /target:Rebuild || exit /b
+ %MSBUILD% msvc\libusb_static_%YEAR%.vcxproj /p:Platform=x64 /p:Configuration=%CONFIG% /target:Rebuild || exit /b

mkdir ..\libusb\include\libusb-1.0
copy libusb\libusb.h ..\libusb\include\libusb-1.0
mkdir ..\libusb\MS64\dll
- copy x64\%CONFIG%\dll\*.lib ..\libusb\MS64\dll
+ copy x64\%CONFIG%\lib\*.lib ..\libusb\MS64\dll
copy x64\%CONFIG%\dll\*.dll ..\libusb\MS64\dll
copy x64\%CONFIG%\dll\*.pdb ..\libusb\MS64\dll

```

That is, add the line to also build the static target (both are needed), and copy the .lib files from \lib rather than \dll.

xlz, 2018-03-18:
> Checkout a specific revision of the libusb github, rather than grabbing from head

It was grabbing from head because we were shipping a private patch. It will after libusb 1.0.22.

> libusb_static_2015.vcproj

We don't support building static libusb. If libusb-1.0.lib was not generated it's a bug then. 

In your log `E:\GitHub\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.lib` is a parameter of link.exe so it's probably generated somewhere. I'll see how this can be fixed when I have time.

closed 2018-11-18 by xlz