OpenKinect/libfreenect2 #977: Erorr :Could not find LibUSB

HUNGLiquid, 2018-03-22:
Overview Description:
Hi, everyone. I'm a college student in Vietnam. 
When i generate your project by Cmake ,i have a error. Can you support me ? Please. 

Actual Results:
Selecting Windows SDK version  to target Windows 10.0.16299.
The C compiler identification is MSVC 19.0.24210.0
The CXX compiler identification is MSVC 19.0.24210.0
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
using tinythread as threading library
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
CMake Error at C:/Program Files/CMake/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find LibUSB (missing: LibUSB_LIBRARIES LibUSB_INCLUDE_DIRS)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindLibUSB.cmake:79 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:90 (FIND_PACKAGE)

Configuring incomplete, errors occurred!
See also "E:/Code/libfreenect2-master/build/CMakeFiles/CMakeOutput.log".

wujita01, 2018-03-24:
me too.HOW TO SOLVE IT

SirDifferential, 2018-03-28:
Please read the [Readme](https://github.com/OpenKinect/libfreenect2/blob/master/README.md) which explains how to build libfreenect2. You haven't prepared the dependencies.

HUNGLiquid, 2018-03-29:
SirDifferential . File README just want to Install libusbK driver and run install_libusb_vs2013.cmd . I done that work but it cant build. I try to download libusb many version but it still like that.

SirDifferential, 2018-03-29:
Could you include the errors you get when you build libusb?

HUNGLiquid, 2018-03-29:
When i run install_libusb_vs2013.cmd it just run fast and close cmd.

SirDifferential, 2018-03-29:
You'll have to run those commands from a git capable command prompt that also has access to msbuild. You can use Git Shell from https://desktop.github.com/

HUNGLiquid, 2018-03-29:
SirDifferential Now i can find libusb but cant find TurboJPEG_LIBRARIES . I edit new enviroment variable TurboJPEG_ROOT with C:\libjpeg-turbo-gcc . What is my mistake? I do same with glfw.

SirDifferential, 2018-03-29:
If you're on Windows, you'll want the msvc version of libjpeg-turbo. You said "libjpeg-turbo-gcc" which means you either got the version that's compiled for GCC on Linux or the mingw version.

You say you're a college student - it might be good practise to compile libjpeg-turbo yourself: https://github.com/libjpeg-turbo/libjpeg-turbo/blob/master/BUILDING.md

HUNGLiquid, 2018-03-30:
SirDifferential Thanks you so much .