OpenKinect/libfreenect2 #978: kinect2 sensor start/stop leakage

vaidotaspeckys, 2018-03-28:
Setup: Operating system win10, usbdk version 1.0.19, sensor kinect2

Way to reproduce, leaks few MB per cycle:
```
	while (true)
	{
		libfreenect2::OpenGLPacketPipeline* foo = new libfreenect2::OpenGLPacketPipeline();
		m_pFreenectDev = m_rFreenect2Ctx.openDevice(m_szSerial, foo);
		std::this_thread::sleep_for(std::chrono::milliseconds(1000));
		if (!m_pFreenectDev)
		{
			continue;
		}

		m_pFreenectListener = new libfreenect2::SyncMultiFrameListener(libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
		
		m_pFreenectDev->setIrAndDepthFrameListener(m_pFreenectListener);

		if (!m_pFreenectDev->start())
		{
			continue;
		}

		m_pFreenectDev->stop();
		m_pFreenectDev->close();

		delete m_pFreenectDev;

		delete m_pFreenectListener;
	}
```

But in oposite, code below doesn't leak(differs type of listiner):
```
while (true)
	{
		libfreenect2::OpenGLPacketPipeline* foo = new libfreenect2::OpenGLPacketPipeline();
		m_pFreenectDev = m_rFreenect2Ctx.openDevice(m_szSerial, foo);
		std::this_thread::sleep_for(std::chrono::milliseconds(1000));

		if (!m_pFreenectDev)
		{
			continue;
		}

		m_pFreenectListener = new libfreenect2::SyncMultiFrameListener(libfreenect2::Frame::Color);
		
		m_pFreenectDev->setColorFrameListener(m_pFreenectListener);

		if (!m_pFreenectDev->start())
		{
			continue;
		}

		m_pFreenectDev->stop();
		m_pFreenectDev->close();

		delete m_pFreenectDev;

		delete m_pFreenectListener;
	}
```

vaidotaspeckys, 2018-03-28:
libfreenect2 version 0.2.0

SirDifferential, 2018-03-28:
You have several large problems in your code.

* You're allocating the packet pipeline and frame listeners on heap. Then `continue` can be called without them being freed.
* Why are you doing "first run" initialization in a loop that never exits? Only allocate the pipeline once, only open the device once.
* What is the purpose of your loop? You aren't reading any data from the sensor, simply allocating some types required for starting the sensor.

Please look at the [Protonect example](https://github.com/OpenKinect/libfreenect2/blob/master/examples/Protonect.cpp) to get an idea on what a typical pipeline looks like.

vaidotaspeckys, 2018-03-28:
Yes, you are right, my code is only for reproducing a leak. The initial issue was rising usage of memory of my application(application for grabbing depth/ir/color) when sensor are started/stoped multiple times.
These code snippets was the minimal piece of code, when leakage are notable.

Pipeline is not freed by intension, because according to openDevice documentation pipeline is freed automatically

```
  /** Open device by serial number.
   * @param serial Serial number
   *** @param factory New PacketPipeline instance. This is always automatically freed.**
   * @return New device object, or NULL on failure
   */
  Freenect2Device *openDevice(const std::string &serial, const PacketPipeline *factory);
```

SirDifferential, 2018-03-29:
I did some testing with this. I can confirm that test case 1 seems to leak memory, test case 2 not, according to valgrind.

The leaks seem to be mostly in the radeonsi_dri.so, AMD driver on Debian 9. The points where valgrind is able to give a line in libfreenect2 are:

libfreenect2::ShaderProgram::build() (opengl_depth_packet_processor.cpp:192)
```
    gl()->glCompileShader(fragment_shader);
```

libfreenect2::ShaderProgram::build() (opengl_depth_packet_processor.cpp:202)
```
    program = gl()->glCreateProgram();
```

libfreenect2::ShaderProgram::build() (opengl_depth_packet_processor.cpp:208)
```
      gl()->glBindFragDataLocation(program, it->second, it->first.c_str());
```

libfreenect2::ShaderProgram::build() (opengl_depth_packet_processor.cpp:211)
```
    gl()->glLinkProgram(program);
```

libfreenect2::ShaderProgram::setVertexShader(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (opengl_depth_packet_processor.cpp:159)
```
    vertex_shader = gl()->glCreateShader(GL_VERTEX_SHADER);
```

libfreenect2::OpenGLDepthPacketProcessorImpl::run(libfreenect2::Frame**, libfreenect2::Frame**) (opengl_depth_packet_processor.cpp:729)
```
    glDrawArrays(GL_TRIANGLES, 0, 6);
```

libfreenect2::OpenGLDepthPacketProcessorImpl::run(libfreenect2::Frame**, libfreenect2::Frame**) (opengl_depth_packet_processor.cpp:756)
```
    gl()->glBindFramebuffer(GL_DRAW_FRAMEBUFFER, stage2_framebuffer);
```

 libfreenect2::OpenGLDepthPacketProcessor::OpenGLDepthPacketProcessor(void*, bool) (opengl_depth_packet_processor.cpp:881)
```
  GLFWwindow* window = glfwCreateWindow(1024, 848, "OpenGLDepthPacketProcessor", 0, parent_window);
```
libfreenect2::Texture<libfreenect2::ImageFormat<2ul, 33332u, 36244u, 5123u> >::allocate(unsigned long, unsigned long) (opengl_depth_packet_processor.cpp:339)
```
    glTexImage2D(GL_TEXTURE_RECTANGLE, 0, FormatT::InternalFormat, width, height, 0, FormatT::Format, FormatT::Type, 0);
```

xlz, 2018-04-14:
I was aware of the 'leaks' in OpenGL and OpenCL libraries when I was previously hunting for them, but it's not clear to me how to stop them, like, there's no glDestroyProgram?