OpenKinect/libfreenect2 #980: I'm using libfreenect2 library with Windows 10. Even though I successfully build the solution and was able to run the Protonect demo, when I include the related directories on my VS15 application I get an error saying:

gregorsamsa183, 2018-03-30:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

SirDifferential, 2018-04-12:
The DLL is loaded upon the executable starting. You must put the DLL somewhere where it can be found - most commonly in the same directory as the executable. If you're starting the program through Visual Studio (or some other IDE) they may have a different working directory and the DLL may be placed somewhere else.

SirDifferential, 2018-04-12:
You can use [Dependency Walker](http://www.dependencywalker.com/) to open the executable and see where this DLL is being searched. It should give an idea on where the DLL is expected to be.

gregorsamsa183, 2018-04-18:
I'm already working with Ubuntu and everything is working fine by setting up the pylibfreenect2 library. For now, I won't be using VS.

closed 2018-11-18 by xlz