OpenKinect/libfreenect2 #983: libusb failling to build

AAcquier, 2018-04-04:
Hello I am on windows 10 and I am trying to install libfreenect 2 when I tried to installed libusb I got this result

~\Documents\GitHub\libfreenect2-master\depends [master +1 ~0 -0 | +55 ~1 -0 !]> .\install_libusb_vs2015.cmd

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends>rem This can only be run in a Git Shell or similar environments

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends>rem with access to git.exe and msbuild.exe.

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends>rmdir /s /q libusb_src libusb
The system cannot find the file specified.

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends>git clone https://github.com/libusb/libusb.git libusb_src   || exit /b
Cloning into 'libusb_src'...
remote: Counting objects: 13040, done.
remote: Compressing objects: 100% (50/50), done.
remote: Total 13040 (delta 103), reused 117 (delta 87), pack-reused 12888
Receiving objects: 100% (13040/13040), 3.49 MiB | 2.95 MiB/s, done.
Resolving deltas: 100% (9435/9435), done.

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends>cd libusb_src

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src>set CONFIG=Release

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src>set YEAR=2015

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src>set MSBUILD="C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe"

C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src>"C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe" msvc\libusb_dll_2015.vcxproj /p:Platform=x64 /p:Configuration=Release /target:Rebuild   || exit /b
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 04/04/2018 16:23:06.
Project "C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\libusb_dll_2015.vcxproj" on node 1
 (Rebuild target(s)).
CoreClean:
  Creating directory "C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\..\x64\Release\dll\li
  busb-1.0 (dll)\".
PrepareForBuild:
  Creating directory "C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\..\x64\Release\dll\li
  busb-1.0 (dll)\libusb-1.0 (dll).tlog\".
InitializeBuildStatus:
  Creating "C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (
  dll)\libusb-1.0 (dll).tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
ClCompile:
  CL.exe /c /I. /I..\libusb /Zi /nologo /W4 /WX- /O2 /GL /D NDEBUG /D WINVER=0x0501 /D _WIN32_WINNT=0x0501 /D _LIB /D _
  CRT_SECURE_NO_WARNINGS /D _WINDLL /D _UNICODE /D UNICODE /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:
  inline /Fo"C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0
  (dll)\\" /Fd"C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.
  0 (dll)\vc140.pdb" /Gd /TC /errorReport:queue ..\libusb\core.c ..\libusb\descriptor.c ..\libusb\hotplug.c ..\libusb\i
  o.c ..\libusb\os\poll_windows.c ..\libusb\strerror.c ..\libusb\sync.c ..\libusb\os\threads_windows.c ..\libusb\os\win
  dows_nt_common.c ..\libusb\os\windows_usbdk.c ..\libusb\os\windows_winusb.c
TRACKER : error TRK0005: Failed to locate: "CL.exe". The system cannot find the file specified. [C:\Users\Owner\Documen
ts\GitHub\libfreenect2-master\depends\libusb_src\msvc\libusb_dll_2015.vcxproj]

Done Building Project "C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\libusb_dll_2015.vcxp
roj" (Rebuild target(s)) -- FAILED.

Build FAILED.

"C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\libusb_dll_2015.vcxproj" (Rebuild target)
(1) ->
(ClCompile target) ->
  TRACKER : error TRK0005: Failed to locate: "CL.exe". The system cannot find the file specified. [C:\Users\Owner\Docum
ents\GitHub\libfreenect2-master\depends\libusb_src\msvc\libusb_dll_2015.vcxproj]

    0 Warning(s)
    1 Error(s)

I check on internet and the result was that I was missing a component in visual basic which I installed but still get exactly the same results. 

Could you help me with that?

SirDifferential, 2018-04-12:
>  TRACKER : error TRK0005: Failed to locate: "CL.exe". The system cannot find the file specified. [C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src\msvc\libusb_dll_2015.vcxproj]

This is suggesting your compiler wasn't found. Do you have Visual Studio 2015 installed?

AAcquier, 2018-04-12:
No but, I got VS2017 installed but manage to install the 2015 compilers (64 and 32 bits) which should have done the trick

SirDifferential, 2018-04-12:
Then you'll have to edit the libusb build script to use version 2017. Change the variables:

```
set YEAR=2015
set MSBUILD="C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe"
```

to

```
set YEAR=2017
set MSBUILD="C:\Program Files (x86)\MSBuild\15.0\Bin\MSBuild.exe"
```

I'm not 100% sure the msbuild path is right - find it yourself and verify it's there.

AAcquier, 2018-04-12:
Thanks, I will try that later , I am working on something else at the moment. I will keep you posted.

AAcquier, 2018-04-13:
I cannot find the build script, any idea which one it is?

SirDifferential, 2018-04-13:
It's [depends/install_libusb_vs2015.cmd](https://github.com/OpenKinect/libfreenect2/blob/master/depends/install_libusb_vs2015.cmd)

AAcquier, 2018-04-13:
Now, I get this error:
C:\Users\Owner\Documents\GitHub\libfreenect2-master\depends\libusb_src>"C:\Program Files (x86)\MSBuild\15.0\Bin\MSBuild.exe" msvc\libusb_dll_2017.vcxproj /p:Platform=x64 /p:Configuration=Release /target:Rebuild   || exit /b
The system cannot find the path specified.

There is no bin folder in the path "C:\Program Files (x86)\MSBuild\15.0

AAcquier, 2018-04-13:
Problem solved, some component of VS2015 were not updated properly, I finally manage to get a build

closed 2018-04-13 by AAcquier

zirop, 2018-11-17:
Could you tell me how I solved this problem in the same situation?
I do not have a bin folder
I am also using Visual Studio 2017

AAcquier, 2018-11-18:
Did you install VS2015, this does not support VS2017

amozeng, 2019-08-15:
Hi,
I am having the same problem:
`TRACKER : error TRK0005: Failed to locate: "CL.exe". The system cannot find the file specified. `
Can you tell me how you fixed this?
I have both VS2017 and 2015 installed.
Thank you so much!

kaisark, 2020-10-25:
Compiled with MS Visual Studio 2015 and Windows SDK 8.1.

---------------------------------
PS C:\libfreenect2\depends> .\install_libusb_vs2015.cmd

C:\libfreenect2\depends>rem This can only be run in a Git Shell or similar environments

C:\libfreenect2\depends>rem with access to git.exe and msbuild.exe.

C:\libfreenect2\depends>rmdir /s /q libusb_src libusb
The system cannot find the file specified.

C:\libfreenect2\depends>git clone https://github.com/libusb/libusb.git libusb_src   || exit /b
Cloning into 'libusb_src'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 15214 (delta 0), reused 2 (delta 0), pack-reused 15205
Receiving objects: 100% (15214/15214), 4.54 MiB | 1.27 MiB/s, done.
Resolving deltas: 100% (11035/11035), done.

C:\libfreenect2\depends>cd libusb_src

C:\libfreenect2\depends\libusb_src>set CONFIG=Release

C:\libfreenect2\depends\libusb_src>set YEAR=2015

C:\libfreenect2\depends\libusb_src>set MSBUILD="C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe"

C:\libfreenect2\depends\libusb_src>"C:\Program Files (x86)\MSBuild\14.0\Bin\MSBuild.exe" msvc\libusb_dll_2015.vcxproj /p:Platform=x64 /p:Configuration=Release /target:Rebuild   || exit /b
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 10/25/2020 11:55:39 AM.
Project "C:\libfreenect2\depends\libusb_src\msvc\libusb_dll_2015.vcxproj" on node 1 (Rebuild target(s)).
CoreClean:
  Creating directory "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\".
PrepareForBuild:
  Creating directory "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0 (dll).tlog\".
InitializeBuildStatus:
  Creating "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0 (dll).tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\CL.exe /c /I. /I..\libusb /Zi /nologo /W4 /WX- /O2 /GL /D NDEBUG /D _WIN32_WINNT=_WIN32_WINNT_VISTA /D _LIB /D _CRT_SECURE_NO_WARNINGS /D _WINDLL /D _UNICODE /D UNIC
  ODE /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\\" /Fd"C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\
  vc140.pdb" /Gd /TC /errorReport:queue /utf-8 ..\libusb\core.c ..\libusb\descriptor.c ..\libusb\os\events_windows.c ..\libusb\hotplug.c ..\libusb\io.c ..\libusb\strerror.c ..\libusb\sync.c ..\libusb\os\threads_windows.c ..\libusb\os\wi
  ndows_common.c ..\libusb\os\windows_usbdk.c ..\libusb\os\windows_winusb.c
  core.c
  descriptor.c
  events_windows.c
  hotplug.c
  io.c
  strerror.c
  sync.c
  threads_windows.c
  windows_common.c
..\libusb\os\windows_common.c(117): warning C4267: '+=': conversion from 'size_t' to 'UINT', possible loss of data [C:\libfreenect2\depends\libusb_src\msvc\libusb_dll_2015.vcxproj]
  windows_usbdk.c
  windows_winusb.c
ResourceCompile:
  C:\Program Files (x86)\Windows Kits\8.1\bin\x86\rc.exe /D _UNICODE /D UNICODE /l"0x0409" /nologo /fo"C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0.res" ..\libusb\libusb-1.0.rc
Link:
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.dll" /NOLOGO kernel32.lib kernel32.lib user32.lib gdi32.lib w
  inspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /DEF:"..\libusb\libusb-1.0.def" /ASSEMBLYRESOURCE:"libusb-1.0.rc" /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /m
  anifest:embed /DEBUG /PDB:"C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.pdb" /LTCG:incremental /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0.lib
  " /MACHINE:X64 /DLL "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\libusb-1.0.res"
  "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\core.obj"
  "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\descriptor.obj"
  "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\events_windows.obj"
  "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\hotplug.obj"
  "C:\libfreenect2\depends\libusb_src\msvc\..\x64\Release\dll\libusb-1.0 (dll)\io.obj"
[... 21 lines cut ...]
  ..\libusb\os\windows_common.c(117): warning C4267: '+=': conversion from 'size_t' to 'UINT', possible loss of data [C:\libfreenect2\depends\libusb_src\msvc\libusb_dll_2015.vcxproj]

    1 Warning(s)
    0 Error(s)

Time Elapsed 00:00:12.14

C:\libfreenect2\depends\libusb_src>mkdir ..\libusb\include\libusb-1.0

C:\libfreenect2\depends\libusb_src>copy libusb\libusb.h ..\libusb\include\libusb-1.0
        1 file(s) copied.

C:\libfreenect2\depends\libusb_src>mkdir ..\libusb\MS64\dll

C:\libfreenect2\depends\libusb_src>copy x64\Release\dll\*.lib ..\libusb\MS64\dll
x64\Release\dll\libusb-1.0.lib
        1 file(s) copied.

C:\libfreenect2\depends\libusb_src>copy x64\Release\dll\*.dll ..\libusb\MS64\dll
x64\Release\dll\libusb-1.0.dll
        1 file(s) copied.

C:\libfreenect2\depends\libusb_src>copy x64\Release\dll\*.pdb ..\libusb\MS64\dll
x64\Release\dll\libusb-1.0.pdb
        1 file(s) copied.