OpenKinect/libfreenect2 #984: Unable to build on Ubuntu 17.10

laurimiller, 2018-04-09:
Running the cmake command produces the following output:

```
ubuntu@ubuntu:~/Desktop/libfree/libfreenect2/build$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
-- using tinythread as threading library
-- Checking for modules 'libva;libva-drm'
--   No package 'libva' found
--   No package 'libva-drm' found
-- Could NOT find TegraJPEG (missing: TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
CMake Error at /usr/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find TurboJPEG (missing: TurboJPEG_INCLUDE_DIRS TURBOJPEG_WORKS)
Call Stack (most recent call first):
  /usr/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:237 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/Desktop/libfree/libfreenect2/build/CMakeFiles/CMakeOutput.log".
```

I would be ever so grateful for any help!

SirDifferential, 2018-04-10:
```
  Could NOT find TurboJPEG (missing: TurboJPEG_INCLUDE_DIRS TURBOJPEG_WORKS)
```

Do you have libjpeg62-turbo-dev installed?

mamachengcheng, 2018-04-12:
I met the same problem when I build on Ubuntu 17.10.Do you solve it now?

floe, 2018-04-12:
Let me reiterate: did you already execute `sudo apt-get install libturbojpeg libjpeg-turbo8-dev`?

mamachengcheng, 2018-04-12:
Yes, I did it.

floe, 2018-04-12:
And you are getting the exact same error as above? Did you already remove and re-create the `build` directory?

mamachengcheng, 2018-04-12:
I got the same error and I've tried it many times.

floe, 2018-04-16:
What do you see when you run `dpkg -L libturbojpeg` and `dpkg -L libjpeg-turbo8-dev`?

mamachengcheng, 2018-04-16:
When I input these, I can see:
`mcc@mcc-pc:~$ dpkg -L libturbojpeg
/.
/usr
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libturbojpeg.so.0.1.0
/usr/share
/usr/share/doc
/usr/share/doc/libturbojpeg
/usr/share/doc/libturbojpeg/changelog.Debian.gz
/usr/share/doc/libturbojpeg/copyright
/usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
mcc@mcc-pc:~$ dpkg -L libjpeg-turbo8-dev
/.
/usr
/usr/include
/usr/include/jerror.h
/usr/include/jmorecfg.h
/usr/include/jpegint.h
/usr/include/jpeglib.h
/usr/include/x86_64-linux-gnu
/usr/include/x86_64-linux-gnu/jconfig.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libjpeg.a
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libjpeg.pc
/usr/share
/usr/share/doc
/usr/share/doc/libjpeg-turbo8-dev
/usr/share/doc/libjpeg-turbo8-dev/README.ijg.gz
/usr/share/doc/libjpeg-turbo8-dev/README.md.gz
/usr/share/doc/libjpeg-turbo8-dev/copyright
/usr/share/doc/libjpeg-turbo8-dev/libjpeg.txt.gz
/usr/share/doc/libjpeg-turbo8-dev/structure.txt.gz
/usr/lib/x86_64-linux-gnu/libjpeg.so
/usr/share/doc/libjpeg-turbo8-dev/changelog.Debian.gz
`
Thank you for your help!

xlz, 2018-04-17:
Use `apt-get install libturbojpeg0-dev`.

commit 6c8ceed2 (2018-04-17): docs: Update TurboJPEG instructions on Ubuntu

turbojpeg.h, libturbojpeg.{a,so} are moved from libjpeg-turbo8-dev in xenial to libturbojpeg0-dev in artful.

Fix #984.
closed 2018-04-17 by xlz