OpenKinect/libfreenect2 #987: No data in Depth and IR frames data arrays

KriStanev, 2018-04-17:
I made a simple clone of the Protonect project to use for testing occlusion- I only need to draw the colour frame to screen and I will be using the depth data to determine occlusion. Whenever I get the frames from the kinect, the colour frame is fine and full of data to use. However, the depth and Ir frames data arrays are empty.

I am using an OpenGLPacketPipeline but I tried all of them and encounter the same problem with all of the pipelines- colour data is present in the frame and ready to use, depth and Ir data is not accessible from the frame map. 

My setup is fine and everything is as it should be. According to the API I am taking the proper steps to access the data, but the depth frames still come in with an empty data array.

I spent time debugging the protonect example to look at the step by step execution and even though all frames are being rendered to the window and obviously the data is coming through and being used and drawn, the frame map in the protonect example during debug also shows empty data arrays for the depth and Ir frames. Colour frame data array is fine.

What am I missing?

SirDifferential, 2018-04-18:
I think it's the best if you simply share your code so we can get an idea on if there's a problem in the code or if this is a USB problem or something like that.

KriStanev, 2018-04-18:
Firstly, thank you for the willingness to help.

I am attaching my two classes that handle the Kinect. All includes are as they should be and everything works fine theoretically. It is really that one issue where I am trying to directly access the data from the FrameMap. The FrameMap has 3 frames in it on every update. The Colour frame's data array is full of data and I can successfully copy it to my own texture and use it to render with OpenGL.
The data in the Depth frame simply stays empty always, nothing to copy from it.

[Dropbox.zip](https://github.com/OpenKinect/libfreenect2/files/1923994/Dropbox.zip)

The Kinect itself works absolutely fine! Here is the weird thing:
I tried creating the OpenGLPacketPipeline in two ways:
1. Creating my own glfw window and context first and passing it to the constructor of the OpenGLPacketPipeline. What this does is it creates two windows- my own as well as the PacketPipeline window, which DRAWS the frames. Obviously data is being transferred, used and drawn implicitly, which is not what I need.

2. Creating the OpenGLPacketPipeline first (or simply not passing in my own GLFWWindow)- This only creates my own window and nothing is being drawn to it (because I haven't set up my draw calls properly, which will happen at a later stage).

No matter how I create the packet pipeline, Colour data always comes in and depth and Ir data is always absent in the frames. 

I am also attaching some screenshots.

[image]

here in debug mode I am having a look at the colour frame and its data array is full 

[image]

Depth and Ir frame's array is always empty.

[image]

And here is what the application looks like when creating the pipeline the first way mentioned above, just as proof that otherwise everything seems to be set up and working properly. 

P.S. I know I have a lot of packet loss. I would guess this is due to either crappy USB or crappy Graphics card. I am running it on a bad laptop. The application still manages to provide output despite that.

P.P.S. My OpenGL draw calls in the Texture.cpp are not properly set up, but this is not important at this stage until I have the data acquisition work properly. I am going to read into OpenGL rendering and texturing after. Apologies if it makes you cringe

KriStanev, 2018-04-18:
Tried again just in case- using the CPU pipeline provides the same result:
Colour data array is full
Ir and Depth- empty.

closed 2018-11-18 by xlz